ros-tooling/cross_compile #19: New cross-compile workflow implementation

prajakta-gokhale, 2019-08-07:
@ros2/aws-robotics is working on a proposed new design for the cross-compile workflow (ros2/design#233 and ros2/ros2#693) and we will be opening PRs in this repo for our changes.

@filiperinaldi would you be open to reviewing our PRs? If you or any other maintainer of this package would like to discuss the design, I can set up a meeting to talk about it.

Since this repo is not maintained by OSRF, would it be possible for members of our team to get write access to it in the future?

pokitoz, 2019-08-08:
Hello!

> @filiperinaldi would you be open to reviewing our PRs? If you or any other maintainer of this package would like to discuss the design, I can set up a meeting to talk about it.

We are keeping an eye on this new design and it looks promising :)

> Since this repo is not maintained by OSRF, would it be possible for members of our team to get write access to it in the future?

For now, @filiperinaldi is on holiday until end of August. Unfortunately, I cannot answer to this question and wanted to let you know that the answers for this issue will be delayed.

emersonknapp, 2019-10-17:
The first version of this new workflow has been implemented and checked in with https://github.com/ros-tooling/cross_compile/pull/22

closed 2019-10-17 by emersonknapp