ros-tooling/cross_compile #32: Question about the DDS implementation used in this project

left4taco, 2019-10-22:
Hi there,
It's exciting to see a whole new cross compile script. We are running into problem when doing cross compile in the old fashion with another DDS implementation(i.e., fastrtps is disabled). 

I'm wondering when you test the cross compile script in this repository, are you using FastRTPS or RTI Connext Pro?

Thanks.

prajakta-gokhale, 2019-10-22:
Hi @left4taco! Currently we are using FastRTPS in the script and have not tested any other DDS implementations.

What is the error that you are getting while trying to use the old version of the script?

left4taco, 2019-10-22:
@prajakta-gokhale Thanks. The problem that we saw is kinda complicated, and I'll try to make understandable. 

We developed our own RMW for RTI Connext Micro, based on the the RTI Connext Pro package created by OSRF(https://github.com/ros2/rosidl_typesupport_connext). 

We want to disable FastRTPS and build with RTI Connext Micro only. But the cross compilation failed on `rcl` package. Errors are like this:

```
/usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu
/bin/ld: warning: librcl_interfaces__rosidl_typesupport_apex_dds_c.so, 
needed by /home/hao.peng/gc/apex_ws/install_xcompile/rcl_interfaces/lib/
librcl_interfaces__rosidl_typesupport_c.so, not found (try using -rpath or -rpath-link)

/usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu
/bin/ld: warning: libbuiltin_interfaces__rosidl_typesupport_apex_dds_cpp.so
, needed by /home/hao.peng/gc/apex_ws/install_xcompile/rcl_interfaces/lib/
librcl_interfaces__rosidl_typesupport_c.so, not found (try using -rpath or -rpath-link)
```

Basically it says, the generic `librcl_interfaces__rosidl_typesupport_c.so` cannot find the RMW specific shared library `librcl_interfaces__rosidl_typesupport_apex_dds_c.so`, though whose path is already in the `rpath` list. 

Interestingly, natively compiling on an aarch64 machine with RTI Connext Micro only succeeded without errors.

After some debugging, I found that if I add these paths(`install/rcl_interfaces/lib`) into `--rpath-link` argument, the cross-compiling linker is then able to find the file. 

My only assumption is that the cross-compiling linker `/usr/lib/gcc-cross/aarch64-linux-gnu/7/collect2` does not search for needed libraries in `rpath`, but rather in `rpath-link`, which is reasonable because according to its document,
```
  -rpath PATH                 Set runtime shared library search path
  -rpath-link PATH            Set link time shared library search path
```

What I don't understand is that why the `x86_64` compiler(the native compiler) is able to find the msg shared libraries in `rpath` while the cross compiler can't. Could you share some of your thoughts of what might be happening here and how to make cross_compiler also search in `rpath`?

Thanks!

---

The reason that cross compile doesn't fail for FastRTPS is that FastRTPS uses `ament_export_dependencies` to add the RMW specific msg libraries to `rcl_interfaces_LIBRARIES`, which will be explicitly added to the compiler command. Therefore, symbols can be found there. 
However, the `RTI_Connext_Pro` cmake file exports using `rosidl_export_typesupport_libraries` [here](https://github.com/ros2/rosidl_typesupport_connext/blob/master/rosidl_typesupport_connext_cpp/cmake/rosidl_typesupport_connext_cpp_generate_interfaces.cmake#L238)

prajakta-gokhale, 2019-10-22:
Thanks for the info!

It is a little hard to point out what could be wrong without looking at the source and workspace setup. A good starting point could be going through the CMake documentation on RPATH that describes build flags and their default values:  https://gitlab.kitware.com/cmake/community/wikis/doc/cmake/RPATH-handling.

If you want to try out the new tool for your DDS implementation we would be happy to assist in that. It will be easier for us to debug issues that way as well and having RTI Connext supported in the tool would be useful.

left4taco, 2019-10-23:
@prajakta-gokhale Thanks! That's helpful. 

I actually also found a thread from CMake mailing list that explains exactly what's happening. 

https://cmake.org/pipermail/cmake-developers/2018-July/030746.html

prajakta-gokhale, 2019-10-23:
Great! Feel free to reopen this issue if you'd like to discuss anything else.

closed 2019-10-23 by prajakta-gokhale