ros-tooling/cross_compile #38: Sysroot mixin should be referenced using file:// not http://

thomas-moulard, 2019-10-29:
The following [line](https://github.com/ros-tooling/cross_compile/blob/d9fdad12f8a9ef79f544dbb0eaeacdc5cdb5cba2/cross_compile/Dockerfile_workspace#L97-L98) seems wrong to me:

```
RUN colcon mixin add cc_mixin \
    https://raw.githubusercontent.com/ros-tooling/cross_compile/master/mixins/index.yaml
```

Could we copy the mixin and use instead:

```
RUN colcon mixin add cc_mixin \
    file://.../mixins/index.yaml
```

The  main issue with the current approach is that anyone developing on a feature branch will end up using the mixin from master. So it will be impossible to change the mixin locally w/o changing the Dockerfile.

Generally, it also open the door to other subtle issues where the script and the mixin aren't in sync because they pull from different locations.

thomas-moulard, 2019-10-29:
@piraka9011 maybe that could be tackled as part as your oncall?

piraka9011, 2019-10-30:
The reason we do this is because we copy the `Dockerfile_workspace` to the `sysroot` directory where all the packages are cross compiled. We don't know where the mixins will be located relative to the `sysroot` directory and so we can't copy them to the Docker image.

We can add this functionality and it will involve the following:
1. Add a command to clone the `cross_compile` in the `Dockerfile_workspace`.
2. Point `colcon mixin add` to the mixin in the cloned repo.

thomas-moulard, 2019-10-30:
Well you don't want to clone right, otherwise you're back to step 1 (is it the right branch?).
Could we copy the mixin in the user workspace in this case, for instance?

piraka9011, 2019-10-30:
> Could we copy the mixin in the user workspace in this case, for instance? Or pass the mixin path as a Dockerfile param?

We can do this as well, so copying the mixin to the sysroot and then copying into the Docker image.

zmichaels11, 2020-01-03:
@emersonknapp update with in progress work

emersonknapp, 2020-01-07:
There isn't work in progress on this - after reviewing I have retitled and rewritten the issue description to describe the actual problem and solution.

commit 506bea15 (2020-01-07): WIP use local copy of mixins for the colcon build

Fixes #38
commit 407d12b9 (2020-01-07): WIP use local copy of mixins for the colcon build

Fixes #38
commit 177445d9 (2020-01-07): Use local copy of mixins for the colcon build, instead of master

Fixes #38
commit 1157fb9b (2020-01-08): Use local copy of mixins for the colcon build, instead of master

Fixes #38
closed 2020-01-08 by emersonknapp