ros-tooling/cross_compile #41: cross_compile problem on sysroot_compile.py

JoelinChee, 2019-10-30:
Hello, everyone! Sorry to bother you. I have one question when I cross compile ros2.

when I Install prerequisites and Installing from source. However, when I do 

colcon build --packages-up-to cross_compile, 

Here come a problem, 

Starting >>> cross_compile
--- stderr: cross_compile                    
  File "/home/pibotvm/ros2_cross_compile_ws/install/cross_compile/lib/python3.5/site-packages/cross_compile/sysroot_compiler.py", line 84
    SYSROOT_DIR_NAME: str = 'sysroot'
                    ^
SyntaxError: invalid syntax

---
Finished <<< cross_compile [3.32s]

Does anyone know how to solve this problem? 

THX.

deitry, 2019-10-30:
Possibly it is due to Python 3.5 version. What kind of OS and Python you are using?

On Kubuntu 18.04, which I am using, Python 3.6 installed from official repos handles this correctly.

piraka9011, 2019-10-30:
Syntax for variable annotations were not added until Python 3.6 (See [PEP-526](https://www.python.org/dev/peps/pep-0526/)). You are running 3.5.

I recommend you upgrade your Python version to 3.6+ to use this.

JoelinChee, 2019-10-31:
THX deitry, THX piraka9011. I have changed Python 3.5 to Python 3.6. And It works.