ros-tooling/cross_compile #42: cross_compile: pkg_resources.DistributionNotFound 'docker'

potentialdiffer, 2019-10-30:
Hey people,
I am trying out the new toolchain on Debian Stretch but running into an issue:
```
$ ros2 run cross_compile cross_compile --sysroot-base-image raspbian/stretch --sysroot-path /path/to/sysroot/

Traceback (most recent call last):
  File "/home/xxx/ros2/install/cross_compile/lib/cross_compile/cross_compile", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/xxx/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 3251, in <module>
    @_call_aside
  File "/home/xxx/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/home/xxx/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/xxx/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/home/xxx/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/xxx/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'docker' distribution was not found and is required by cross-compile
```
docker-ce is installed and runs without any problems. Might this be due to Debian Stretch python version? 
Thanks!

potentialdiffer, 2019-10-30:
I tried with python 3.6 but I am still receiving the same error message.

thomas-moulard, 2019-10-31:
Thanks for the bug report @potentialdiffer.
How are you installing cross_compile? Using the Debian package, or from source?

potentialdiffer, 2019-10-31:
I am building all my workspaces from source and therefore cross_compile as well.

zmichaels11, 2019-10-31:
I'm going to try and reproduce your issue.
I'm trying:
* ROS2 distro: dashing from source
* OS: debian stretch

zmichaels11, 2019-10-31:
I was able to reproduce this error. 

A workaround is to install the python docker package directly with pip.
`pip3 install docker`

emersonknapp, 2019-11-01:
@potentialdiffer did you run `rosdep install` against your workspace to install the runtime dependencies? `python3-docker` is a runtime dependency of this package, as specified in the package.xml, and it does have a rule in rosdistro for debian. https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L5037 - it would be good to know if that's not working properly

zmichaels11, 2019-11-01:
I just tried `rosdep install` and it did try and install `python3-docker`, however the version of `python3-docker` in Debian stretch is too old to run cross_compile. (Stretch has python3-docker 1.9.0-1)

potentialdiffer, 2019-11-05:
I did `rosdep install`:
```
rosdep install -r --rosdistro=dashing --ignore-packages-from-source --from-paths src/
```
I get some errors due to other packages. But I resoveld those before by installing them manually. Bo error occurs due to `cross_compile` package.
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rmw_fastrtps_dynamic_cpp: No definition of [fastrtps] for OS [debian]
urdfdom: No definition of [urdfdom_headers] for OS [debian]
kdl_parser: No definition of [urdfdom_headers] for OS [debian]
rosidl_typesupport_connext_cpp: No definition of [rti-connext-dds-5.3.1] for OS [debian]
rosidl_typesupport_opensplice_cpp: No definition of [libopensplice69] for OS [debian]
robot_state_publisher: No definition of [urdfdom_headers] for OS [debian]
rosidl_typesupport_fastrtps_c: No definition of [fastrtps] for OS [debian]
connext_cmake_module: No definition of [rti-connext-dds-5.3.1] for OS [debian]
rmw_connext_cpp: No definition of [rti-connext-dds-5.3.1] for OS [debian]
rosidl_typesupport_opensplice_c: No definition of [libopensplice69] for OS [debian]
rmw_connext_shared_cpp: No definition of [rti-connext-dds-5.3.1] for OS [debian]
opensplice_cmake_module: No definition of [libopensplice69] for OS [debian]
rmw_opensplice_cpp: No definition of [libopensplice69] for OS [debian]
rosidl_typesupport_fastrtps_cpp: No definition of [fastrtps] for OS [debian]
urdf: No definition of [urdfdom_headers] for OS [debian]
rmw_fastrtps_cpp: No definition of [fastrtps] for OS [debian]
rmw_fastrtps_shared_cpp: No definition of [fastrtps] for OS [debian]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

Installing `python3-docker` with apt did not solve the problem, however installing with pip, as @zmichaels11 mentioned fixed this issue for me.

As mentioned in #41 I needed to install python 3.6 in advance to get it all work.

I am not sure what the initial problem was. Could it be that `rosdep` did not install the docker dependency as it was the wrong python version?

zmichaels11, 2019-11-05:
I'm looking into having rosdep install python3-docker from pip.

zmichaels11, 2019-11-08:
Update:
`python3-docker` should be installed by pip only on Debian 9