ros-tooling/cross_compile #43: ros2: command not found

JoelinChee, 2019-10-31:
Hi everyone. 

My OS is ubuntu 16.04.

I have successfully cross compile by using:
colcon build --packages-up-to cross_compile

and I have successfully source the local_setup.sh by using:
source install/local_setup.sh

However, when I Copy the QEMU Binaries and Copy the ROS2 packages, I have problems on 

pibotvm@ubuntu:~$ ros2 run cross_compile cross_compile --arch aarch64 --os ubuntu --sysroot-path /home/pibotvm/sysroot/
ros2: command not found

May is there anything wrong?

JoelinChee, 2019-10-31:
I have installed ros1.0  kinetic in my computer

deitry, 2019-10-31:
You must have any ROS2 distrbution installed, like `ros-dashing-desktop` metapackage. And yet you have to do `source /opt/ros/dashing/setup.sh` before running any ros2 command.