ros-tooling/cross_compile #48: Document in README.md which Python versions are supported

thomas-moulard, 2019-11-01:
I.e. the version for which we run the CI.

zmichaels11, 2019-11-01:
Note: Python 3.5 does work if typing on variables is removed. ([PEP 526](https://www.python.org/dev/peps/pep-0526/); 3.6+ feature) 
We should determine if we need to support Python 3.5.

thomas-moulard, 2019-11-01:
We don't need to make this decision, the REP-2000 is there for that:
https://www.ros.org/reps/rep-2000.html#dashing-diademata-may-2019-may-2021

Dashing -> Minimum language requirements: Python 3.5

But fixing the code to actually support the right version of Python is out of the scope of this ticket IMHO. We should start by documenting what we expect.

zmichaels11, 2019-11-08:
As of PR #53, this project depends on Python ≥ 3.5

closed 2019-11-18 by prajakta-gokhale