ros-tooling/cross_compile #50: Docker multi-stage build can be used to circumvent copying qemu-user-static

zmichaels11, 2019-11-01:
[Docker Documentation](https://docs.docker.com/develop/develop-images/multistage-build/#use-multi-stage-builds)

Instead of 
```Dockerfile
COPY qemu-user-static/ /usr/bin/
```

Something like:
```Dockerfile
FROM ubuntu:bionic as qemu

RUN apt install -y qemu-user-static

ARG ROS2_BASE_IMG
FROM ${ROS2_BASE_IMG}
ARG ROS2_WORKSPACE
ARG ROS_DISTRO
ARG TARGET_TRIPLE
ARG TARGET_ARCH

COPY --from=qemu /usr/bin/qemu-* /usr/bin/
```

zmichaels11, 2019-11-01:
[WIP branch](https://github.com/ros-tooling/cross_compile/tree/multi-stage-docker)

The changes appear to work if a dummy file is placed in `qemu-user-static` folder.
From sysroot, run `touch qemu-user-static/dummy`

Changes required:
- [ ] Remove checks on `qemu-user-static` folder
- [ ] Apply changes to `Dockerfile_workspace`
- [ ] Update documentation

thomas-moulard, 2019-11-06:
Docker can't run ARM binaries, the support comes from a Linux kernel feature called binfmt_misc.
When the kernel is asked to start running a binary, if it's not a native binary, it checks to see if it can forward the execution request to a user space program which will "interpret" the binary for the kernel.

When we install qemu-user-static, we depend on qemu-user which registers the following binfmt_misc hooks:

```
$ ls /proc/sys/fs/binfmt_misc
jar  llvm-6.0-runtime.binfmt  python2.7  python3.6  qemu-aarch64  qemu-alpha  qemu-arm  qemu-armeb  qemu-cris  qemu-m68k  qemu-microblaze  qemu-mips  qemu-mips64  qemu-mips64el  qemu-mipsel  qemu-ppc  qemu-ppc64  qemu-ppc64abi32  qemu-ppc64le  qemu-s390x  qemu-sh4  qemu-sh4eb  qemu-sparc  qemu-sparc32plus  qemu-sparc64  register  sbcl  status
```

```
$ cat /proc/sys/fs/binfmt_misc/qemu-aarch64
enabled
interpreter /usr/bin/qemu-aarch64-static
flags: OC
offset 0
magic 7f454c460201010000000000000000000200b700
mask ffffffffffffff00fffffffffffffffffeffffff
```

What it means, is that every time a cross_compile ARM binary is executed, the execution is actually handled by `/usr/bin/qemu-aarch64-static`​, which is why you need to copy those in the Docker image.
The beauty of this design is that Docker does not need to support running non-x86 images on x86.

This also means that we can't strip the qemu binaries from our Docker containers, as it would break the 

More here: https://ownyourbits.com/2018/06/13/transparently-running-binaries-from-any-architecture-in-linux-with-qemu-and-binfmt_misc/

thomas-moulard, 2019-11-11:
@zmichaels11 I don't think we can implement this idea, as we need qemu binaries at runtime, please re-open if you think otherwise.

closed 2019-11-11 by thomas-moulard

zmichaels11, 2019-11-11:
I think its probably going to be better to mount `/usr/bin/qemu-aarch64-static` instead of copying it in the `Dockerfile`.