ros-tooling/cross_compile #62: `export_workspace_sysroot_image` takes too much time

deitry, 2019-11-08:
For reference:
https://github.com/ros-tooling/cross_compile/blob/5fa1f16e96e5002ae2abdf189f46389788864d55/cross_compile/sysroot_compiler.py#L293

Export takes much more time than build itself. On my PC build time of packages up to `rcl` takes about 15-20 mins and export took more than half-hour before I drop it - while `docker cp` directly from resulting image lasts for seconds. 

As far as I can see, this is happening because export process itself is single-threaded, though I didn't noticed what actually takes so long.

piraka9011, 2019-11-08:
Once assets are compiled in the Dockerfile, we remove previous assets in the sysroot and replace them with the cross-compiled binaries and libraries.
Exporting is a docker-py specifc task so there's not much we can do to optimize there to the best of my knowledge.

piraka9011, 2019-11-12:
Resolved by #63

closed 2019-11-12 by piraka9011