ros-tooling/cross_compile #67: Create a GH Action that runs cross-compile

zmichaels11, 2019-11-21:
The cross-compile workflow can be simplified by having a GH Action run cross-compile and upload the workspace as an artifact.

thomas-moulard, 2019-11-21:
We already have an e2e test. We just need to enable it I guess?

zmichaels11, 2019-11-21:
This isn't adding an e2e test, its creating a new github action for the marketplace to run `cross_compile` as a CICD step.

thomas-moulard, 2019-11-21:
Oh I get what you're saying now, how do you upload the ws, seems too big to be an artefact, does that incorporate pushing the docker image to the GitHub registry?

zmichaels11, 2019-11-21:
[GitHub is cool as long as its no more than 2GB](https://help.github.com/en/github/managing-large-files/distributing-large-binaries)