ros-tooling/cross_compile #80: Debian build fail.

jfm92, 2019-12-11:
Hi, 
I'm trying to cross-compile ROS2 for the RPI3 but it fail when is installling the ROS2 dependencies. This is the log:
``` bash
 ros2 run cross_compile cross_compile --sysroot-path ~/ros2_cc_pck  -a armhf -o debian
INFO:cross_compile.sysroot_compiler:Checking sysroot directory... foo
INFO:cross_compile.sysroot_compiler:Fetching sysroot base image: arm32v7/debian:latest
INFO:cross_compile.sysroot_compiler:Building workspace image: juan/armhf-debian-fastrtps-dashing:latest
INFO:cross_compile.sysroot_compiler:Step 1/29 : ARG ROS2_BASE_IMG
INFO:cross_compile.sysroot_compiler:Step 2/29 : FROM ${ROS2_BASE_IMG}
INFO:cross_compile.sysroot_compiler:---> b6efee0ab4f5
INFO:cross_compile.sysroot_compiler:Step 3/29 : ARG ROS2_WORKSPACE
INFO:cross_compile.sysroot_compiler:---> Using cache
INFO:cross_compile.sysroot_compiler:---> b4d6b4fe6f8a
INFO:cross_compile.sysroot_compiler:Step 4/29 : ARG ROS_DISTRO
INFO:cross_compile.sysroot_compiler:---> Using cache
INFO:cross_compile.sysroot_compiler:---> 1c504fef1e15
INFO:cross_compile.sysroot_compiler:Step 5/29 : ARG TARGET_TRIPLE
INFO:cross_compile.sysroot_compiler:---> Using cache
INFO:cross_compile.sysroot_compiler:---> 79d35faa8a1e
INFO:cross_compile.sysroot_compiler:Step 6/29 : ARG TARGET_ARCH
INFO:cross_compile.sysroot_compiler:---> Using cache
INFO:cross_compile.sysroot_compiler:---> e153e7b67843
INFO:cross_compile.sysroot_compiler:Step 7/29 : COPY qemu-user-static/ /usr/bin/
INFO:cross_compile.sysroot_compiler:---> Using cache
INFO:cross_compile.sysroot_compiler:---> eb6cd45fd965
INFO:cross_compile.sysroot_compiler:Step 8/29 : RUN echo 'Etc/UTC' > /etc/timezone &&     ln -sf /usr/share/zoneinfo/Etc/UTC /etc/localtime
INFO:cross_compile.sysroot_compiler:---> Using cache
INFO:cross_compile.sysroot_compiler:---> 4c9649feeeaf
INFO:cross_compile.sysroot_compiler:Step 9/29 : RUN apt-get update && apt-get install -y         tzdata         locales     && rm -rf /var/lib/apt/lists/*
INFO:cross_compile.sysroot_compiler:---> Using cache
INFO:cross_compile.sysroot_compiler:---> 42ab1e05224f
INFO:cross_compile.sysroot_compiler:Step 10/29 : RUN locale-gen en_US en_US.UTF-8 &&     update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
INFO:cross_compile.sysroot_compiler:---> Running in 8378b349d6e7
INFO:cross_compile.sysroot_compiler:Generating locales (this might take a while)...
INFO:cross_compile.sysroot_compiler:Generation complete.
INFO:cross_compile.sysroot_compiler:*** update-locale: Error: invalid locale settings:  LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8

ERROR:cross_compile.sysroot_compiler:Error building sysroot image. The following error was caught:
The command '/bin/sh -c locale-gen en_US en_US.UTF-8 &&     update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8' returned a non-zero code: 255
NoneType: None
ERROR:cross_compile.sysroot_compiler:BuildError does not take keyword arguments
Traceback (most recent call last):
  File "/home/juan/ros2_cc_pck/install/cross_compile/lib/python3.6/site-packages/cross_compile/sysroot_compiler.py", line 287, in execute_cc_pipeline
    self.build_workspace_sysroot_image()
  File "/home/juan/ros2_cc_pck/install/cross_compile/lib/python3.6/site-packages/cross_compile/sysroot_compiler.py", line 251, in build_workspace_sysroot_image
    raise docker.errors.BuildError(reason=error_line, build_log=error_line)
TypeError: BuildError does not take keyword arguments
INFO:cross_compile.ros2_cross_compile:To setup the cross compilation build environment:
1. Run the command below to setup using sysroot's GLIBC for cross-compilation.
`bash .`
2. Run the command below to export the environment variables used by the cross-compiled ROS packages.
`source .`'
```
If I execute ``locate`` inside of the docker, it returns that is POSIX instead of UTF-8, I don't know if this might be the problem.

I'm working with Ubuntu 18.04.

zmichaels11, 2019-12-11:
Hi jfm92,

I noticed that you're using `debian` for your cross compile target.
Cross Compile currently only supports `ubuntu` for target image.

I have opened up an issue #81 to add support for Debian as a target.

I looked a bit into why this is failing:
on a fresh Debian image, `/etc/locale.gen` has `en_US.UTF-8` commented out. 
I'll open up a PR to properly generate the `en_US.UTF-8` locale.

zmichaels11, 2019-12-12:
Hi @jfm92 ,

Can you try this again using `master`?
The following PRs were merged in:
* #82 fixes the locale-gen error you encountered
* #85 fixes later problems that arose when running `./test/run_e2e_test.sh --os debian`

jfm92, 2019-12-18:
I just tried and now it works, thank you!

closed 2019-12-18 by jfm92