ros-tooling/cross_compile #101: Cross compile  issues with own compiler toolchains and sysroot

zihui0325, 2019-12-30:
I want to cross compile ros2 to Hi3559 platform with the compiler toolchains and sysroot that Hisi provide. This is the detail : https://github.com/ros2/rclcpp/issues/957#issue-543104847 

I cross compiled the deps `apr`, `apr-util`, `log4cxx`, `tinyxml`, `tinyxml2`, `poco`, `openssl`, `python3.6` in advance and install them to the target filesystem.

Here is the TOOLCHAIN file.
TOOLCHAIN_FILE
```
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSTEM_PROCESSOR aarch64)

set(HISI_ROOT "/opt/hisi-linux/x86-arm/aarch64-himix100-linux")
set(CMAKE_C_COMPILER "${HISI_ROOT}/bin/aarch64-himix100-linux-gcc")
set(CMAKE_CXX_COMPILER "${HISI_ROOT}/bin/aarch64-himix100-linux-g++")

# This is the target sysroot
set(CMAKE_SYSROOT "${HISI_ROOT}/target")

set(CMAKE_FIND_ROOT_PATH
    # This is where I install ros2
    /home/r18119/ros_hisi/dashing/install
    # This is where I cross compiled deps install
    /home/r18119/hisi_target
)

MESSAGE(STATUS ${CMAKE_FIND_ROOT_PATH})

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

# This assumes that pthread will be available on the target system
    # (this emulates that the return of the TRY_RUN is a return code "0"
set(THREADS_PTHREAD_ARG "0"
      CACHE STRING "Result from TRY_RUN" FORCE)
```

There are two problems when cross compiling `tf2_ros` and `rclcpp_components` package, which I think can be sorted to one kind.

The following are err logs:
`tf2_ros`
```
--- stderr: tf2_ros                                                                                                               
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../lib/gcc/aarch64-linux-gnu/6.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: librosidl_typesupport_fastrtps_cpp.so, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../lib/gcc/aarch64-linux-gnu/6.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: librosidl_typesupport_fastrtps_c.so, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../lib/gcc/aarch64-linux-gnu/6.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: libfastrtps.so.1, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../lib/gcc/aarch64-linux-gnu/6.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: libssl.so.3, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../lib/gcc/aarch64-linux-gnu/6.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: libcrypto.so.3, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../lib/gcc/aarch64-linux-gnu/6.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: libfastcdr.so.1, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../lib/gcc/aarch64-linux-gnu/6.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: libPocoFoundation.so.64, needed by /home/r18119/ros_hisi/dashing/install/lib/librosidl_typesupport_c.so, not found (try using -rpath or -rpath-link)
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../lib/gcc/aarch64-linux-gnu/6.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: libyaml.so, needed by /home/r18119/ros_hisi/dashing/install/lib/librcl_yaml_param_parser.so, not found (try using -rpath or -rpath-link)
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserialize(char&)'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserialize(short&)'
/home/r18119/ros_hisi/dashing/install/lib/librcl_interfaces__rosidl_typesupport_fastrtps_cpp.so: undefined reference to `eprosima::fastcdr::Cdr::serializeBoolSequence(std::vector<bool, std::allocator<bool> > const&)'
/home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so: undefined reference to `rosidl_typesupport_fastrtps_c__identifier'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::serialize(double)'
/home/r18119/ros_hisi/dashing/install/lib/librosidl_typesupport_c.so: undefined reference to `Poco::SharedLibrary::hasSymbol(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserializeArray(float*, unsigned long)'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserializeArray(int*, unsigned long)'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so: undefined reference to `eprosima::fastcdr::Cdr::state::state(eprosima::fastcdr::Cdr const&)'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserializeArray(char*, unsigned long)'
/home/r18119/ros_hisi/dashing/install/lib/librcl_yaml_param_parser.so: undefined reference to `yaml_parser_delete'
/home/r18119/ros_hisi/dashing/install/lib/librcl_interfaces__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::serializeArray(bool const*, unsigned long)'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserializeArray(double*, unsigned long)'
/home/r18119/ros_hisi/dashing/install/lib/librcl_yaml_param_parser.so: undefined reference to `yaml_parser_set_input_file'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::serializeArray(long const*, unsigned long)'
/home/r18119/ros_hisi/dashing/install/lib/librcl_yaml_param_parser.so: undefined reference to `yaml_event_delete'
/home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::serialize(int)'
/home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserialize(int&)'
/home/r18119/ros_hisi/dashing/install/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserializeArray(short*, unsigned long)'
[... 60 lines cut ...]
/home/r18119/ros_hisi/dashing/install/lib/librcl_interfaces__rosidl_typesupport_fastrtps_cpp.so: undefined reference to `eprosima::fastcdr::Cdr::setState(eprosima::fastcdr::Cdr::state&)'
/home/r18119/ros_hisi/dashing/install/lib/librcl_interfaces__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::serialize(long)'
/home/r18119/ros_hisi/dashing/install/lib/librcl_interfaces__rosidl_typesupport_fastrtps_cpp.so: undefined reference to `rosidl_typesupport_fastrtps_cpp::typesupport_identifier'
/home/r18119/ros_hisi/dashing/install/lib/librcl_interfaces__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::deserializeArray(long*, unsigned long)'
/home/r18119/ros_hisi/dashing/install/lib/librcl_interfaces__rosidl_typesupport_fastrtps_c.so: undefined reference to `eprosima::fastcdr::Cdr::serializeArray(double const*, unsigned long)'
/home/r18119/ros_hisi/dashing/install/lib/librcl_interfaces__rosidl_typesupport_fastrtps_cpp.so: undefined reference to `typeinfo for eprosima::fastcdr::exception::Exception'
[... cut ...]
make[2]: *** [component_container_mt] Error 1
make[1]: *** [CMakeFiles/component_container_mt.dir/all] Error 2
make: *** [all] Error 2
---

```
It seems like that it can not find the libraries depend on when building `librcl_interfaces__*.so`, so I add 'rpath' to cmake toolchain file , but it still doesn't work. 
```
# /home/r18119/ros_hisi/dashing/install is where I install ros2
add_compile_options(-L/home/r18119/ros_hisi/dashing/install/lib -Wl,-rpath=/home/r18119/ros_hisi/dashing/install)
```

zihui0325, 2020-01-02:
I checked that `rclcpp_component` and `tf2_ros`  packages both depend on `rclcpp` package with `ament_target_dependencies(target rclcpp)`, but actually it doesn't link to libraries on which 'rclcpp' denpend recursively. 

After I add  `rmw_fastrtps_cpp` and `yaml` to `ros2/rclcpp/rclcpp_components/CMakeLists.txt` manually as follows,  it worked.
```
target_link_libraries(component_container component_manager)
 ament_target_dependencies(component_container
   "rclcpp"
   "rmw_fastrtps_cpp"
   "yaml"
 )
```
However,  I can build native from source successfully. There may be a problem with the `ament_target_dependencies()` when cross compling ?

emersonknapp, 2020-01-02:
This repository contains a tool that performs a cross-compilation automatically. Unfortunately, the specifics of your setup are not within the scope of this tool. If you have specific feature requests for the tool, such as "allow custom toolchain file / sysroot / etc", we would very much welcome those as feature request issues. However, since this here seems specific to your manual use case, I am going to close this issue as off-topic. I would recommend perhaps https://answers.ros.org/questions/ or https://stackoverflow.com/ for this question.

closed 2020-01-02 by emersonknapp