ros-tooling/cross_compile #132: Codecov CI action fails on PRs from forks

emersonknapp, 2020-02-04:
## Description
When a Pull Request comes in from a fork, it can't access the secret codecov token, so the CI fails to upload codecov results and returns a failure. This was observed here: https://github.com/ros-tooling/cross_compile/pull/131

Latest news is that there is a GitHub Actions API released today, and the codecov action is working to implement a solution - see https://github.com/codecov/codecov-action/issues/29

For now, this issue can track that progress.

### Expected Behavior
* PRs from forks can upload codecov results

### Actual Behavior
* PRs from forks fail CI on the tasks that try to upload to codecov

## To Reproduce
* Open a PR from a fork of this repo

## System (please complete the following information)
N/A

## Additional context
N/A

prajakta-gokhale, 2020-02-22:
This is true for all repos using (action-ros-ci)[https://github.com/ros-tooling/action-ros-ci]. Similar issue: https://github.com/ros-tooling/action-ros-ci/issues/56.

emersonknapp, 2020-02-22:
According to https://github.com/codecov/codecov-action/issues/29#issuecomment-589353189 we should have some option available in the next week or so.

emersonknapp, 2020-03-05:
Codecov action "tokenless uploads" have been enabled - check if it works

emersonknapp, 2020-03-10:
This is now working - see report on https://github.com/ros-tooling/cross_compile/pull/171 - accurately reported reduced coverage

closed 2020-03-10 by emersonknapp