ros-tooling/cross_compile #133: Release package as debs for Debian/Ubuntu installation

emersonknapp, 2020-02-04:
## Description

Other Python packages in the ROS ecosystem are available both via `pip` and `apt`, e.g. `colcon`. Follow their lead and release this package into `.deb`s also.

## Related Issues

N/A

## Completion Criteria

A user can `apt install ros-cross-compile` from the OpenRobotics apt repositories.

## Implementation Notes / Suggestions

Follow the instructions https://github.com/ros-infrastructure/ros_release_python here to verify packaging, and send request to OSRF to add the package to the build farm for upload.

## Testing Notes / Suggestions

Generate the deb locally and try installing it into a fresh environment. Perhaps use `rocker` since it allows for docker-from-docker running.

thomas-moulard, 2020-02-04:
My two cents: it's preferable to only have one way to do "things". If we start having a second way to release `cross_compile`, we're basically doubling the surface we need to cover with our canaries, which will be a maintenance cost for us. I'd suggest not to proceed with this unless we can articulate a gain for our users which would justify the increased maintenance cost.

emersonknapp, 2020-02-04:
The following is the argument in favor

> we almost always recommend using the debs for users as it auto upgrades. There's many times a pip install gets overlayed, but when a new version comes out via Debian, if a new dependency got updated, it will start crashing because they are accidentally using the pip overlay with the updated dependencies that are no longer compatible.
> It should probably just require adding an stdeb.cfg file and other than that the overhead is low.

zmichaels11, 2020-05-28:
Not planning to work on this in the immediate future.

closed 2020-05-28 by zmichaels11