ros-tooling/cross_compile #135: Allow for custom SKIP_ROSDEP_KEYS

Timple, 2020-02-04:
## Description

We have rosdep keys that cannot be resolved easily and are unimportant for the compilation of the package. They will be present on the target system

## Related Issues

None

## Completion Criteria

Optional skip-rosdep argument, or an environment variable that can be set in the custom-setup-script

## Implementation Notes / Suggestions

Add a second `--skip-keys` argument to the `rosdep install` command with a user specified variable 
or
Add to the SKIP_ROSDEP_KEYS environment variable that already exists

## Testing Notes / Suggestions

Add a non-resolvable dependency to a test-package and ignore it (or not) with the new option.

emersonknapp, 2020-02-04:
Confirmed useful, moving to backlog. Not covered by the `--custom-setup-script` case since it can't export environment (and the var would be overwritten anyways). Going to plan to add this after the current rework for #106  is done - unless somebody else picks it up.

> unimportant for the compilation of the package

That makes me think these are `run_depends`. As a workaround for the moment, you could comment out those lines in the `package.xml`.

> cannot be resolved easily

If it's the case that you have custom rosdep rules that need installed, you can use the `--custom-setup-script` to enable them (if they were necessary for the build), see https://github.com/ros-tooling/cross_compile#custom-setup-script

dabonnie, 2020-03-13:
Assigned to Snehal

PR #180 (2020-04-07): Allow user to skip rosdep keys.
Fixes #135.

This feature helps user to skip the rosdep keys that are not important for the compilation of package.
closed 2020-04-07 by emersonknapp