ros-tooling/cross_compile #201: Error library linking libtf2.so from libtf2_ros.so ??

mce16, 2020-05-20:
## Description
** 
i followed the instructions here (installed docker etc..)

https://index.ros.org/doc/ros2/Tutorials/Cross-compilation/#cross-compiling-examples-for-arm
and got what seemed to be the library linking error
 **

### Expected Behavior
** successfull built **

### Actual Behavior
** 
[172.192s] /usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu/bin/ld: warning: libtf2.so, needed by /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so, not found (try using -rpath or -rpath-link)
[172.193s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::canTransform(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*) const'
[172.195s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::canTransform(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*) const'
[172.196s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::~BufferCore()'
[172.196s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `typeinfo for tf2::BufferCore'
[172.198s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::lookupTransform(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
[172.201s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::clear()'
[172.202s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::setTransform(geometry_msgs::msg::TransformStamped_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
[172.203s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::allFramesAsYAML[abi:cxx11]() const'
[172.203s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::BufferCore(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)'
[172.203s] /home/kk/cc_ws/ros2_ws/install/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::lookupTransform(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&) const'
[172.205s] collect2: error: ld returned 1 exit status

 **

## To Reproduce
** 
colcon build --merge-install \
    --cmake-force-configure \
    --cmake-args \
        -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON \
        -DCMAKE_TOOLCHAIN_FILE="$(pwd)/src/ros2/cross_compile/cmake-toolchains/generic_linux.cmake" \
        -DSECURITY=ON

 **

## System (please complete the following information)
 - OS: [e.g. Ubuntu Bionic]

**  **

## Additional context
** Add any other context about the problem here **

piraka9011, 2020-05-20:
Are you using the legacy instructions?
Can you use the new instructions documented in the `README.md`

mce16, 2020-05-21:
I have the docker and qemu installed previously following the "legacy" instructions; 

But here i  was simply lost trying to follow the instructions here; after the line
pip3 install ros_cross_compile; 

what do i do next? Invoking ros_cross_compile just return with command not found.

ros_cross_compile /path/to/my/workspace --arch aarch64 --os ubuntu --rosdistro eloquent

emersonknapp, 2020-05-21:
It sounds like you do not have the pip bin directory on your PATH. On Ubuntu, it is likely to be `~/.local/bin/` for user-installed pip packages. Take a look at that dir and see if it contains `ros_cross_compile` - if so, then add it to `PATH`. Another likely location is `/usr/local/bin`

mce16, 2020-05-26:
Ok, i have exported the path, more errors encountered.. something is missing, but i couldn't figure out what it is

see attached output;

[cc.txt](https://github.com/ros-tooling/cross_compile/files/4679393/cc.txt)

piraka9011, 2020-05-26:
Is the Docker daemon (`dockerd`) running? See the [Docker docs](https://docs.docker.com/config/daemon/) for more info.
Also, is your user part of the `docker` user group?

```
sudo usermod -aG docker `whoami`
```

emersonknapp, 2020-05-26:
Note that I have opened https://github.com/ros-tooling/cross_compile/issues/202 to track an enhancement to fix the need for the above ^ troubleshooting in the future.

mce16, 2020-05-27:
Hi, I am very sure my docker service was running, and i had added myself to the docker group as well as per the installation steps. 

As i see it, it is just one command 
kk@xxxx:~/cc_ws$ ros_cross_compile ros2_ws/ --arch aarch64 --os ubuntu --rosdistro dashing

As i understand it, the command will create the necessary docker image(s) and do  ros2 download and setup inside the image, cross compile to arm etc and all that. I have initially tried to follow the 0.0.1 instructions which lead me to the library linking error as mentioned in the title. 

Btw, I am running ubuntu bionic host on a WSL Windows 10, but i don't think this make any difference anyway.

piraka9011, 2020-05-27:
It sounds like you need to explicitly specify the socket connection for Docker since it listens by default to a Unix socket (something like `/var/run/docker.sock`) but you stated the daemon is exposed on another socket.

Maybe [this SO answer](https://stackoverflow.com/questions/36765138/bind-to-docker-socket-on-windows) might help.

mce16, 2020-05-28:
Hi, i think docker has no problem. It runs correctly and connect to daemon on the Windows 10 host,
via the socket connector for Docker, via  $DOCKER_HOST variable, which points to localhost:2375

It connects to the Daemon in Windows Docker Dekstop without problem, For instance, command docker -info gave me the required info, this command will not run if the docker can't connect to daemon. So on my WSL ubuntu, docker is definitely working correctly.

Is the ros_cross_compile trying to check if the dockerd is running in the WSL? Because i don't think there is any dockerd running in WSL, as Docker daemon cannot run directly on WSL, instead it uses the Docker CLI to connect to a remote Docker daemon running through Docker for Windows. 

As ros_cross_compile is running on WSL, is there anything to skip the docker check if that is the issue? Because i am sure i can build and run docker images in WSL without problem, and my docker is connecting successfully to the docker daemon on Windows host.

emersonknapp, 2020-05-28:
Sorry that we probably won't be able to provide any support with the legacy workflow. It is however a legitimate way of going about cross-compiling ROS code - the intent of this tool is to move towards automating that approach in a user friendly way.

To this issue:

> Is the ros_cross_compile trying to check if the dockerd is running in the WSL, or trying to start the dockerd on WSL ?

`ros_cross_compile` is using the https://docker-py.readthedocs.io/en/stable/ library to invoke Docker - here's how it's currently constructing the low level client https://github.com/ros-tooling/cross_compile/blob/master/ros_cross_compile/docker_client.py#L66 - a naive url to the unix docker.sock

```
docker_api = docker.APIClient(base_url='unix://var/run/docker.sock')
```

We're using the low level `docker.APIClient` to get access to the live output streams to print in real time - looks like the normal `DockerClient.from_env()` explicitly looks at the `DOCKER_HOST` environment variable.

I've opened https://github.com/ros-tooling/cross_compile/pull/204 to try and address this - I'm not sure yet if it works quite right, it would be useful if you could try it out and let me know if it helps with your situation.

mce16, 2020-05-30:
This works.

However, i have another issue on WSL, which seemed to be related to mounting of directory.

see below output:

Command:
kk@xxxx:~/ros2_dashing$ ros_cross_compile $(pwd) --rosdistro dashing --arch armhf --os ubuntu

And yes, my src/ exists in ros2_dashing/

INFO:Docker Client: ---> bca57682f623
INFO:Docker Client:Step 7/10 : COPY gather_rosdeps.sh /root/
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> c6e9b7235fdf
INFO:Docker Client:Step 8/10 : RUN mkdir -p /ws
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 485a72388d4f
INFO:Docker Client:Step 9/10 : WORKDIR /ws
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 0ebda28db54a
INFO:Docker Client:Step 10/10 : ENTRYPOINT ["/root/gather_rosdeps.sh"]
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 8d703e0fd24b
INFO:Docker Client:Successfully built 8d703e0fd24b
INFO:Docker Client:Successfully tagged ros_cross_compile:rosdep
INFO:Rosdep Gatherer:Running rosdep collector image on workspace /home/kk/ros2_dashing
INFO:Docker Client:+ '[' '!' -d ./src ']'
INFO:Docker Client:++ pwd
INFO:Docker Client:+ echo 'No src/ directory found at /ws, did you remember to mount your workspace?'
INFO:Docker Client:+ exit 1
INFO:Docker Client:No src/ directory found at /ws, did you remember to mount your workspace?
Traceback (most recent call last):
  File "/home/kk/.local/bin/ros_cross_compile", line 11, in <module>
    load_entry_point('ros-cross-compile', 'console_scripts', 'ros_cross_compile')()
  File "/home/kk/cc_src/cross_compile/ros_cross_compile/ros_cross_compile.py", line 172, in main
    cross_compile_pipeline(args)
  File "/home/kk/cc_src/cross_compile/ros_cross_compile/ros_cross_compile.py", line 163, in cross_compile_pipeline
    custom_data_dir=custom_data_dir)
  File "/home/kk/cc_src/cross_compile/ros_cross_compile/dependencies.py", line 84, in gather_rosdeps
    volumes=volumes,
  File "/home/kk/cc_src/cross_compile/ros_cross_compile/docker_client.py", line 142, in run_container
    image_name, exit_code, '', image_name, 'See above ^')
docker.errors.ContainerError: Command '' in image 'ros_cross_compile:rosdep' returned non-zero exit status 1: See above ^

emersonknapp, 2020-06-02:
Yes, the script tries to mount the workspace onto the running container, but it appears that maybe it's not being successful. You can try this manual test, which is exactly what's happening under the hood

```
docker run \                                                                                                                                                       
  -v $(pwd):/ws \
  -e OWNER_USER=$(id -u) \
  -e ROSDISTRO=dashing \
  -e TARGET_OS=ubuntu:bionic \
  -e CUSTOM_SETUP='' \
  -e OUT_PATH=/ws/rosdep-test.bash \
  ros_cross_compile:rosdep

# should have created this file
cat rosdep-test.bash
```

And if that doesn't work, perhaps simply

```
docker run -it \                                                                                                                                                       
  -v $(pwd):/ws \
  --entrypoint=/bin/bash \
  ros_cross_compile:rosdep

# now try to list /ws and see what's there... it should be your workspace, src/ install/ build/ etc...
```

emersonknapp, 2020-07-30:
Hello @mce16 did the above test yield any results for you? I've merged https://github.com/ros-tooling/cross_compile/pull/204 which solves at least part of this issue - I will close this ticket for now, please reopen it if you still have something we need to work through. Thanks!

closed 2020-07-30 by emersonknapp