ros-tooling/cross_compile #203: Error issue from rosdep during ros_cross_compile from Virtualbox ubuntu bionic 18.04

mce16, 2020-05-28:
As per error output, it said no /src directory found;
What confused me were:
I don't think i need to create a /src in my workspace? I expect the ros_cross_compile is automated and do all the necessaries?

Secondly, even after i tried to create the src/ in my workspace directory, and delete the docker cache and re-run the command,  the same error encountered.

Below is my one line command, running from virtualbox, ubuntu bionic 18.04; docker installed.

<pre><font color="#8AE234"><b>kk@kk-VirtualBox</b></font>:<font color="#729FCF"><b>~/cc_ws</b></font>$ ros_cross_compile ros2_ws --arch aarch64 --os ubuntu --rosdistro dashing
</pre>

	<pre>	rosdep update
INFO:Docker Client: ---&gt; 11862f5def1c
INFO:Docker Client:Step 7/10 : COPY gather_rosdeps.sh /root/
INFO:Docker Client: ---&gt; 8c9e7f3398eb
INFO:Docker Client:Step 8/10 : RUN mkdir -p /ws
INFO:Docker Client: ---&gt; Running in 20e904cd622f
INFO:Docker Client: ---&gt; b22e62d727d1
INFO:Docker Client:Step 9/10 : WORKDIR /ws
INFO:Docker Client: ---&gt; Running in 65122d66b95c
INFO:Docker Client: ---&gt; dcc15e4b876b
INFO:Docker Client:Step 10/10 : ENTRYPOINT [&quot;/root/gather_rosdeps.sh&quot;]
INFO:Docker Client: ---&gt; Running in 9e942006b978
INFO:Docker Client: ---&gt; dc487cf63ae0
INFO:Docker Client:Successfully built dc487cf63ae0
INFO:Docker Client:Successfully tagged ros_cross_compile:rosdep
INFO:Rosdep Gatherer:Running rosdep collector image on workspace ros2_ws
INFO:Docker Client:No src/ directory found at /ws, did you remember to mount your workspace?
INFO:Docker Client:+ &apos;[&apos; &apos;!&apos; -d ./src &apos;]&apos;
INFO:Docker Client:++ pwd
INFO:Docker Client:+ echo &apos;No src/ directory found at /ws, did you remember to mount your workspace?&apos;
INFO:Docker Client:+ exit 1
Traceback (most recent call last):
  File &quot;/home/kk/.local/bin/ros_cross_compile&quot;, line 11, in &lt;module&gt;
    sys.exit(main())
  File &quot;/home/kk/.local/lib/python3.6/site-packages/ros_cross_compile/ros_cross_compile.py&quot;, line 172, in main
    cross_compile_pipeline(args)
  File &quot;/home/kk/.local/lib/python3.6/site-packages/ros_cross_compile/ros_cross_compile.py&quot;, line 163, in cross_compile_pipeline
    custom_data_dir=custom_data_dir)
  File &quot;/home/kk/.local/lib/python3.6/site-packages/ros_cross_compile/dependencies.py&quot;, line 84, in gather_rosdeps
    volumes=volumes,
  File &quot;/home/kk/.local/lib/python3.6/site-packages/ros_cross_compile/docker_client.py&quot;, line 141, in run_container
    image_name, exit_code, &apos;&apos;, image_name, &apos;See above ^&apos;)
docker.errors.ContainerError: Command &apos;&apos; in image &apos;ros_cross_compile:rosdep&apos; returned non-zero exit status 1: See above ^
</pre>

emersonknapp, 2020-05-28:
Is `~/cc_ws/ros_ws/src/` an existing directory? What happens if you try providing an absolute path, e.g. `ros_cross_compile $(pwd)/ros_ws etc...`?

mce16, 2020-05-28:
No, there is no src/ in ~/cc_ws/ros_ws/

The result is the same if i do the following:
<pre><font color="#8AE234"><b>kk@kk-VirtualBox</b></font>:<font color="#729FCF"><b>~/cc_ws</b></font>$ ros_cross_compile $(pwd)/ros2_ws --arch aarch64 --os ubuntu --rosdistro dashing
</pre>

And now if i were to create a src/ in  ~/cc_ws/ros_ws/; it seems to get passed the error above; but with a new issue pops up like below:

<pre>INFO:Docker Client:rosdep update
INFO:Docker Client:  update the local rosdep database based on the rosdep sources.
INFO:Docker Client:
INFO:Docker Client:rosdep what-needs &lt;rosdeps&gt;...
INFO:Docker Client:  print a list of packages that declare a rosdep on (at least
INFO:Docker Client:  one of) &lt;rosdeps&gt;
INFO:Docker Client:
INFO:Docker Client:rosdep where-defined &lt;rosdeps&gt;...
INFO:Docker Client:  print a list of yaml files that declare a rosdep on (at least
INFO:Docker Client:  one of) &lt;rosdeps&gt;
INFO:Docker Client:
INFO:Docker Client:rosdep fix-permissions
INFO:Docker Client:  Recursively change the permissions of the user&apos;s ros home directory.
INFO:Docker Client:  May require sudo.  Can be useful to fix permissions after calling
INFO:Docker Client:  &quot;rosdep update&quot; with sudo accidentally.
INFO:Docker Client:
INFO:Docker Client:
INFO:Docker Client:rosdep: error: no packages or stacks specified
Traceback (most recent call last):
  File &quot;/home/kk/.local/bin/ros_cross_compile&quot;, line 11, in &lt;module&gt;
    sys.exit(main())
  File &quot;/home/kk/.local/lib/python3.6/site-packages/ros_cross_compile/ros_cross_compile.py&quot;, line 172, in main
    cross_compile_pipeline(args)
  File &quot;/home/kk/.local/lib/python3.6/site-packages/ros_cross_compile/ros_cross_compile.py&quot;, line 163, in cross_compile_pipeline
    custom_data_dir=custom_data_dir)
  File &quot;/home/kk/.local/lib/python3.6/site-packages/ros_cross_compile/dependencies.py&quot;, line 84, in gather_rosdeps
    volumes=volumes,
  File &quot;/home/kk/.local/lib/python3.6/site-packages/ros_cross_compile/docker_client.py&quot;, line 141, in run_container
    image_name, exit_code, &apos;&apos;, image_name, &apos;See above ^&apos;)
docker.errors.ContainerError: Command &apos;&apos; in image &apos;ros_cross_compile:rosdep&apos; returned non-zero exit status 2: See above ^
</pre>

emersonknapp, 2020-05-28:
The one positional argument to the tool is intended to be a workspace that you desire to cross-compile. It needs to be a valid colcon workspace, e.g. it contains a `src/` directory with ros package sources in it. Perhaps there is a different path to the workspace that you intend to build, or maybe a different mismatch of expectation?

mce16, 2020-05-29:
I see. I think as new user, it would be helpful too if this info can be included in the READ-ME as well. I sort of thinking a default ros2 source and packages would be downloaded and use unless specified otherwise, or something to that effect.

Now, i have finally built it successfully in virtualbox, i have a question on its lengthy built time.
On x86 virtualbox, it took me 1.5hrs. And on aarch64, it was a whopping 15+ hours.

See below part of the console output:

<pre>INFO:Docker Client:[Processing: test_rclcpp]
INFO:Docker Client:[Processing: test_rclcpp]
INFO:Docker Client:[Processing: test_rclcpp]
INFO:Docker Client:Finished &lt;&lt;&lt; test_rclcpp [27min 49s]
INFO:Docker Client:Starting &gt;&gt;&gt; tf2_eigen
INFO:Docker Client:[Processing: tf2_eigen]
INFO:Docker Client:[Processing: tf2_eigen]
INFO:Docker Client:Finished &lt;&lt;&lt; tf2_eigen [1min 17s]
INFO:Docker Client:Starting &gt;&gt;&gt; tf2_kdl
INFO:Docker Client:[Processing: tf2_kdl]
INFO:Docker Client:[Processing: tf2_kdl]
INFO:Docker Client:Finished &lt;&lt;&lt; tf2_kdl [1min 9s]
INFO:Docker Client:Starting &gt;&gt;&gt; tf2_sensor_msgs
INFO:Docker Client:Finished &lt;&lt;&lt; tf2_sensor_msgs [22.2s]
INFO:Docker Client:Starting &gt;&gt;&gt; dummy_robot_bringup
INFO:Docker Client:Finished &lt;&lt;&lt; dummy_robot_bringup [18.9s]
INFO:Docker Client:Starting &gt;&gt;&gt; pendulum_control
INFO:Docker Client:[Processing: pendulum_control]
INFO:Docker Client:[Processing: pendulum_control]
INFO:Docker Client:Finished &lt;&lt;&lt; pendulum_control [1min 30s]
INFO:Docker Client:Starting &gt;&gt;&gt; ros1_bridge
INFO:Docker Client:--- stderr: ros1_bridge
INFO:Docker Client:CMake Warning at CMakeLists.txt:34 (message):
INFO:Docker Client:  Failed to find ROS 1 roscpp, skipping...
INFO:Docker Client:
INFO:Docker Client:
INFO:Docker Client:---
INFO:Docker Client:Finished &lt;&lt;&lt; ros1_bridge [24.9s]
INFO:Docker Client:Starting &gt;&gt;&gt; ros2lifecycle
INFO:Docker Client:Finished &lt;&lt;&lt; ros2lifecycle [7.82s]
INFO:Docker Client:Starting &gt;&gt;&gt; lifecycle
INFO:Docker Client:[Processing: lifecycle]
INFO:Docker Client:[Processing: lifecycle]
INFO:Docker Client:Finished &lt;&lt;&lt; lifecycle [1min 23s]
INFO:Docker Client:Starting &gt;&gt;&gt; ros2component
INFO:Docker Client:Finished &lt;&lt;&lt; ros2component [7.97s]
INFO:Docker Client:Starting &gt;&gt;&gt; rqt_action
INFO:Docker Client:Finished &lt;&lt;&lt; rqt_action [7.31s]
INFO:Docker Client:Starting &gt;&gt;&gt; rqt_srv
INFO:Docker Client:Finished &lt;&lt;&lt; rqt_srv [18.4s]
INFO:Docker Client:Starting &gt;&gt;&gt; rosbag2_tests
INFO:Docker Client:[Processing: rosbag2_tests]
INFO:Docker Client:[Processing: rosbag2_tests]
INFO:Docker Client:[Processing: rosbag2_tests]
INFO:Docker Client:[Processing: rosbag2_tests]
INFO:Docker Client:[Processing: rosbag2_tests]
INFO:Docker Client:Finished &lt;&lt;&lt; rosbag2_tests [2min 36s]
INFO:Docker Client:Starting &gt;&gt;&gt; rviz2
INFO:Docker Client:[Processing: rviz2]
INFO:Docker Client:[Processing: rviz2]
INFO:Docker Client:Finished &lt;&lt;&lt; rviz2 [1min 14s]
INFO:Docker Client:
INFO:Docker Client:Summary: 271 packages finished [15h 11min 47s]
INFO:Docker Client:  6 packages had stderr output: cyclonedds rmw_connext_cpp rmw_connext_shared_cpp ros1_bridge rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp
INFO:Docker Client:+ chown -R 1000 .
</pre>

What could be the reason for such a huge (10x) difference? Or is this particular to my setup only?

emersonknapp, 2020-05-29:
The main problem is emulation - it is a lot slower than native builds. We're working on moving to native cross-compilation, which should be more comparable to a native build and take about the same amount of time. This will be a big improvement but probably won't be ready for another couple months.

Regardless, building the entire ros2 desktop installation from source is not really the intended use case - especially not building all the GUI tools which probably shouldn't be used on ARM devices anyways. ROS2 is already built and packaged for aarch64 on ci.ros2.org / build.ros2.org . The real purpose of this tool is to build _your application_, finding dependencies via `package.xml`, and the tool will use the prebuilt packages that already exist for Ubuntu on ARM. If you only have a few packages in your workspace this process will be reasonably fast, especially on subsequent builds - the first run to create the dependency sysroot is going to be a little bit slow.

> it would be helpful too if this info can be included in the READ-ME as well

The README's first sentence is: "A tool to automate compiling ROS and ROS 2 workspaces to non-native architectures.". Is there a way we could make this more clear?

emersonknapp, 2020-07-31:
Closing this in favor of tracking cross-compiler usage in https://github.com/ros-tooling/cross_compile/issues/69

closed 2020-07-31 by emersonknapp