ros-tooling/cross_compile #214: cross compile script fails on OSX

Karsten1987, 2020-06-22:
## Description
I am trying to follow the tutorial and build the [demos](https://github.com/ros2/demos.git) for AARCH64 under OSX. The script unfortunately does not go through and fails with a stacktrace.

### Expected Behavior
Script exits correctly and `build_aarch64` is created.

### Actual Behavior
i am running `ros_cross_compile --arch aarch64 --os ubuntu --rosdistro foxy .`, where `.` is the current workspace.

```
INFO:Rosdep Gatherer:Running rosdep collector image on workspace .
Traceback (most recent call last):
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/docker/api/client.py", line 222, in _raise_for_status
    response.raise_for_status()
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/requests/models.py", line 941, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: Bad Request for url: http+docker://localunixsocket/v1.30/containers/create

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/karsten/.pyenv/versions/3.7.4/bin/ros_cross_compile", line 8, in <module>
    sys.exit(main())
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/ros_cross_compile/ros_cross_compile.py", line 172, in main
    cross_compile_pipeline(args)
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/ros_cross_compile/ros_cross_compile.py", line 163, in cross_compile_pipeline
    custom_data_dir=custom_data_dir)
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/ros_cross_compile/dependencies.py", line 84, in gather_rosdeps
    volumes=volumes,
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/ros_cross_compile/docker_client.py", line 128, in run_container
    network_mode='host',
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/docker/models/containers.py", line 719, in run
    detach=detach, **kwargs)
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/docker/models/containers.py", line 777, in create
    resp = self.client.api.create_container(**create_kwargs)
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/docker/api/container.py", line 450, in create_container
    return self.create_container_from_config(config, name)
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/docker/api/container.py", line 461, in create_container_from_config
    return self._result(res, True)
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/docker/api/client.py", line 228, in _result
    self._raise_for_status(response)
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/docker/api/client.py", line 224, in _raise_for_status
    raise create_api_error_from_http_exception(e)
  File "/Users/karsten/.pyenv/versions/3.7.4/Python.framework/Versions/3.7/lib/python3.7/site-packages/docker/errors.py", line 31, in create_api_error_from_http_exception
    raise cls(e, response=response, explanation=explanation)
docker.errors.APIError: 400 Client Error: Bad Request ("create .: volume name is too short, names should be at least two alphanumeric characters")
```

## System (please complete the following information)
 - OSX
 - ROS 2 Distro: Foxy

emersonknapp, 2020-06-22:
You need to provide a full path, so instead of `.` you should pass `$(pwd)` as the workspace argument. An example can be seen in the tutorial section https://github.com/ros-tooling/cross_compile#running-the-cross-compilation. I can see, however, how this is confusing - it should be fairly trivial to allow relative paths.

emersonknapp, 2020-06-22:
And, the error messaging is completely opaque, so that's a bad experience. Let's keep this issue open to track fixing the usability here.

PR #216 (2020-06-23): Allow relative paths for the workspace argument
Allow relative paths for the workspace argument (via `resolve`), and check early if it looks usable, to prevent waiting until later in the process to raise an issue.

Fixes #214
closed 2020-06-23 by emersonknapp