ros-tooling/cross_compile #244: Skip-Keys: Cannot locate rosdep definition for [console_bridge]

lusim123, 2020-07-24:
## Description
I'm trying to build a cross-compilation for armhf. When I enter the command: 

>  ros_cross_compile -a armhf -d eloquent -o ubuntu $(pwd)

I receive the following error:

> INFO:Docker Client:ERROR: the following packages/stacks could not have their rosdep keys resolved
> INFO:Docker Client:to system dependencies:
> INFO:Docker Client:class_loader: Cannot locate rosdep definition for [console_bridge]
> INFO:Docker Client:urdf: Cannot locate rosdep definition for [console_bridge]
> INFO:Docker Client:urdfdom: Cannot locate rosdep definition for [console_bridge]
> INFO:Docker Client:tf2: Cannot locate rosdep definition for [console_bridge]

My workspace includes a 'src' folder with the ros2.repos and _ros_cross_compile -b 0.5.0_.

For me it seems like something went wrong with the skip-keys, that should include: 

> console_bridge fastcdr fastrtps libopensplice67 libopensplice69 osrf_testing_tools_cpp poco_vendor rmw_connext_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rti-connext-dds-5.3.1 tinyxml_vendor tinyxml2_vendor urdfdom urdfdom_headers

Can somebody reproduce this issue?

lusim123, 2020-07-24:
Issue can be closed.
The problem can be solved by adding the --rosdep_skip_keys option.

closed 2020-07-24 by lusim123

emersonknapp, 2020-07-27:
Glad you figured it out!