ros-tooling/cross_compile #248: What happened with https://github.com/ros-tooling/cross_compile-release.git ?

shr-project, 2020-08-07:
## Description
rosdistro_build_cache is now failing to fetch it, I think it was available before, because I was able to run rosdistro_build_cache with dashing release from 2020-05-29 which already referenced this repo and it didn't fail.

### Expected Behavior
Either the repo should be public or rosdistro should be updated not to reference it.

### Actual Behavior
$ rosdistro_build_cache --debug --preclean --ignore-local $tmpdir/index-v4.yaml dashing
...
DEBUG:rosdistro:Load package.xml file from url "https://raw.githubusercontent.com/rst-tu-dortmund/costmap_converter-ros2-release/release/dashing/costmap_converter_msgs/0.1.0-1/package.xml"
  - updated manifest of package 'costmap_converter_msgs' to version '0.1.0'
  - fetch "costmap_queue"
DEBUG:rosdistro:Load package.xml file from url "https://raw.githubusercontent.com/SteveMacenski/navigation2-release/release/dashing/costmap_queue/0.2.6-1/package.xml"
  - updated manifest of package 'costmap_queue' to version '0.2.6'
  - fetch "cross_compile"
Username for 'https://github.com': 

and fails even if I provide my username and password.

shr-project, 2020-08-07:
https://webcache.googleusercontent.com/search?q=cache:kgaOxZ-YH_sJ:https://github.com/ros-tooling/cross_compile-release+&cd=1&hl=cs&ct=clnk&gl=cz shows that the repo existed before

emersonknapp, 2020-08-07:
Thanks for bringing this up - this was deleted in error as part of an attempt to clean up unused repositories (as cross_compile is not a ros package and does not release to rosdistro any more). We've contacted GitHub support to recover the repository, I'll keep this ticket updated.

emersonknapp, 2020-08-07:
I've restored a snapshot of the repository, let me know if there is any remaining issue that you run in to

shr-project, 2020-08-07:
@emersonknapp thanks for quick update and resolving it promptly. Looks good now.

shr-project, 2020-09-23:
I got it removed from rosdistro with https://github.com/ros/rosdistro/pull/26671, closing this one again.

closed 2020-09-23 by shr-project