ros-tooling/cross_compile #285: Allow caching of Docker Image for reproducible builds.

casmer, 2020-12-31:
## Description

There is no way to specify and save the docker image created for a build to avoid having to redownload and setup the build environment. This also means that if updates occur to the base image and packages in the target image they will get applied not allowing one to build exactly what they build on a particular date.

## Completion Criteria

Provide command line switches to do the following:
- create a tagged/cached image for docker at the specified path, should allow for creating a full image without performing the build.
- use the provided docker image to build, (should no require setup scripts and parameters, as those will not change from last build based on image used.
- provide a mechanism to update an image

## Testing Notes / Suggestions

** create a build environment and output a build, use the cached environment to build an identical build. **

emersonknapp, 2020-12-31:
Good call - you can reuse the exact same image when you carefully don't touch anything, but the moment something gets shifted or a base image updates, you're right, it can take a really long time to rebuild, and not necessarily reproduce the results.

There are a few ways we could approach improving the experience, but I think that definitely custom tags for images to reuse instead of running specific steps would be a great way to go.

kjeremy, 2021-05-25:
This would shave off about 15 minutes from my build... especially in the docker-in-docker case where caching becomes more complicated.

kjeremy, 2021-05-29:
If someone can post a general outline of how to accomplish this I can try to tackle it next week.

emersonknapp, 2021-06-01:
A reasonable-sounding first-pass:

* Add commandline option `--use-build-image TAG`
* If the option is specified, 
  * use it to override `Platform.sysroot_image_tag` - maybe as a `Platform.override_sysroot_image_tag(tag)` method  
  * Add `"gather_rosdeps"` and `"sysroot"` to `stages_skip` in `ros_cross_compile.py` - so that those steps are not run, and we just use the provided tag