ros-tooling/cross_compile #301: rosdep update failed "read operation timed out"

sqn175, 2021-04-13:
## Description

Following the tutorial in [README.md](https://github.com/ros-tooling/cross_compile#tutorial) produces "ERROR: error loading sources list: ('The read operation timed out',)" .

### Expected Behavior
The tutorial should succeed without error.

### Actual Behavior
The error shows as follows:

> INFO:Docker Client:Step 8/10 : RUN mkdir -p /ws
INFO:Docker Client: ---> Running in 72a57723ebed
INFO:Docker Client: ---> 83c01c7c6132
INFO:Docker Client:Step 9/10 : WORKDIR /ws
INFO:Docker Client: ---> Running in 9b6120ff8128
INFO:Docker Client: ---> d8f6798fd20f
INFO:Docker Client:Step 10/10 : ENTRYPOINT ["/root/gather_rosdeps.sh"]
INFO:Docker Client: ---> Running in 83d44075a07f
INFO:Docker Client: ---> d06d4d778f44
INFO:Docker Client:Successfully built d06d4d778f44
INFO:Docker Client:Successfully tagged ros_cross_compile:rosdep
INFO:Rosdep Gatherer:Running rosdep collector image on workspace /home/ubuntu/cross_compile/cross_compile_ws
INFO:Docker Client:+ '[' '!' -d ./src ']'
INFO:Docker Client:+ '[' -f /usercustom/rosdep_setup ']'
INFO:Docker Client:++ dirname cc_internals/aarch64-ubuntu-foxy/install_rosdeps.sh
INFO:Docker Client:+ out_dir=cc_internals/aarch64-ubuntu-foxy
INFO:Docker Client:+ mkdir -p cc_internals/aarch64-ubuntu-foxy
INFO:Docker Client:+ rosdep update --include-eol-distros
INFO:Docker Client:Warning: running 'rosdep update' as root is not recommended.
INFO:Docker Client:  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
INFO:Docker Client:ERROR: error loading sources list:
INFO:Docker Client:     ('The read operation timed out',)
INFO:Docker Client:reading in sources list data from /etc/ros/rosdep/sources.list.d
Traceback (most recent call last):
  File "/home/ubuntu/cross_compile/venv/bin/ros_cross_compile", line 33, in <module>
    sys.exit(load_entry_point('ros-cross-compile', 'console_scripts', 'ros_cross_compile')())
  File "/home/ubuntu/cross_compile/ros_cross_compile/ros_cross_compile.py", line 237, in main
    cross_compile_pipeline(args, data_collector, platform)
  File "/home/ubuntu/cross_compile/ros_cross_compile/ros_cross_compile.py", line 224, in cross_compile_pipeline
    stage(platform, docker_client, ros_workspace_dir, options, data_collector)
  File "/home/ubuntu/cross_compile/ros_cross_compile/dependencies.py", line 136, in __call__
    custom_data_dir=options.custom_data_dir)
  File "/home/ubuntu/cross_compile/ros_cross_compile/dependencies.py", line 87, in gather_rosdeps
    volumes=volumes,
  File "/home/ubuntu/cross_compile/ros_cross_compile/docker_client.py", line 145, in run_container
    image_name, exit_code, '', image_name, 'See above ^')
docker.errors.ContainerError: Command '' in image 'ros_cross_compile:rosdep' returned non-zero exit status 1: See above ^

## To Reproduce
Following the tutorial in [README.md](https://github.com/ros-tooling/cross_compile#tutorial) with command "ros_cross_compile cross_compile_ws/ --rosdistro foxy --arch aarch64 --os ubuntu".

## System (please complete the following information)
 - OS: Ubuntu Bionic
 - ROS 2 Distro: Not installed in host PC

** For ROS 2 bugs - please attach a VCS.repos or Dockerfile if possible to help us reproduce the environment **

## Additional context
I have reproduced this error either in my local host PC or in a AWS cloud server. It looks like a network issue, however I have no idea to solve it. Many thanks!

emersonknapp, 2021-04-13:
I'm not sure - but it sounds at first glance like your docker service is misconfigured and can't reach the network. Could you try the following in the same environment?

```
docker run -it ubuntu:focal
# in the container
apt-get update && apt-get install -y wget
wget google.com
```

If that doesn't succeed, then it suggests your docker service isn't set up to reach the internet. If it works, then it seems like it's something more specific. You could then try

```
docker run -it osrf/ros2:nightly
# in the container
rosdep update
```

Let me know how those experiments go for you.

sqn175, 2021-04-14:
Hi emersonknapp, Thanks for your reply. The "wget google.com" comand failed in both the host PC and the container. This makes sense since google.com is blocked in CHINA mainland. I finally make "rosdep update" works using a proxy network in Docker. I'll close this issue.

closed 2021-04-14 by sqn175

emersonknapp, 2021-04-14:
>  This makes sense since google.com is blocked in CHINA mainland
 
Oh, yes, a better test would be `wget https://raw.githubusercontent.com` - since that's where rosdep pulls its sources from. I am used to `google.com` because it's always available for me - sorry for any confusion there.

I'm glad you figured it out!