ros-tooling/cross_compile #305: Permission denied error when I run the ros_cross_compile command

malith1992, 2021-04-22:
## Description
When I run the `ros_cross_compile /home/nvidia/catkin_ws --arch aarch64 --os ubuntu --rosdistro dashing` command, I get a permission denied error for the sock.connect. 

### Actual Behavior

> nvidia@nvidia:~$ ros_cross_compile /home/nvidia/catkin_ws --arch aarch64 --os ubuntu --rosdistro dashing
> INFO:Rosdep Gatherer:Building rosdep collector image: ros_cross_compile:rosdep
> Traceback (most recent call last):
>   File "/home/nvidia/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 706, in urlopen
>     chunked=chunked,
>   File "/home/nvidia/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 394, in _make_request
>     conn.request(method, url, **httplib_request_kw)
>   File "/usr/lib/python3.6/http/client.py", line 1281, in request
>     self._send_request(method, url, body, headers, encode_chunked)
>   File "/usr/lib/python3.6/http/client.py", line 1327, in _send_request
>     self.endheaders(body, encode_chunked=encode_chunked)
>   File "/usr/lib/python3.6/http/client.py", line 1276, in endheaders
>     self._send_output(message_body, encode_chunked=encode_chunked)
>   File "/usr/lib/python3.6/http/client.py", line 1042, in _send_output
>     self.send(msg)
>   File "/usr/lib/python3.6/http/client.py", line 980, in send
>     self.connect()
>   File "/home/nvidia/.local/lib/python3.6/site-packages/docker/transport/unixconn.py", line 46, in connect
>     sock.connect(self.unix_socket)
> PermissionError: [Errno 13] Permission denied
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/home/nvidia/.local/lib/python3.6/site-packages/requests/adapters.py", line 449, in send
>     timeout=timeout
>   File "/home/nvidia/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 756, in urlopen
>     method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
>   File "/home/nvidia/.local/lib/python3.6/site-packages/urllib3/util/retry.py", line 532, in increment
>     raise six.reraise(type(error), error, _stacktrace)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/urllib3/packages/six.py", line 734, in reraise
>     raise value.with_traceback(tb)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 706, in urlopen
>     chunked=chunked,
>   File "/home/nvidia/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 394, in _make_request
>     conn.request(method, url, **httplib_request_kw)
>   File "/usr/lib/python3.6/http/client.py", line 1281, in request
>     self._send_request(method, url, body, headers, encode_chunked)
>   File "/usr/lib/python3.6/http/client.py", line 1327, in _send_request
>     self.endheaders(body, encode_chunked=encode_chunked)
>   File "/usr/lib/python3.6/http/client.py", line 1276, in endheaders
>     self._send_output(message_body, encode_chunked=encode_chunked)
>   File "/usr/lib/python3.6/http/client.py", line 1042, in _send_output
>     self.send(msg)
>   File "/usr/lib/python3.6/http/client.py", line 980, in send
>     self.connect()
>   File "/home/nvidia/.local/lib/python3.6/site-packages/docker/transport/unixconn.py", line 46, in connect
>     sock.connect(self.unix_socket)
> urllib3.exceptions.ProtocolError: ('Connection aborted.', PermissionError(13, 'Permission denied'))
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/home/nvidia/.local/bin/ros_cross_compile", line 11, in <module>
>     sys.exit(main())
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/ros_cross_compile.py", line 227, in main
>     cross_compile_pipeline(args, data_collector, platform)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/ros_cross_compile.py", line 214, in cross_compile_pipeline
>     stage(platform, docker_client, ros_workspace_dir, options, data_collector)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/dependencies.py", line 136, in __call__
>     custom_data_dir=options.custom_data_dir)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/dependencies.py", line 65, in gather_rosdeps
>     tag=_IMG_NAME,
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/docker_client.py", line 75, in build_image
>     decode=True,
>   File "/home/nvidia/.local/lib/python3.6/site-packages/docker/api/build.py", line 258, in build
>     timeout=timeout,
>   File "/home/nvidia/.local/lib/python3.6/site-packages/docker/utils/decorators.py", line 46, in inner
>     return f(self, *args, **kwargs)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/docker/api/client.py", line 187, in _post
>     return self.post(url, **self._set_request_timeout(kwargs))
>   File "/home/nvidia/.local/lib/python3.6/site-packages/requests/sessions.py", line 590, in post
>     return self.request('POST', url, data=data, json=json, **kwargs)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/requests/sessions.py", line 542, in request
>     resp = self.send(prep, **send_kwargs)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/requests/sessions.py", line 655, in send
>     r = adapter.send(request, **kwargs)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/requests/adapters.py", line 498, in send
>     raise ConnectionError(err, request=request)
> requests.exceptions.ConnectionError: ('Connection aborted.', PermissionError(13, 'Permission denied'))
> 

## To Reproduce
** Steps to reproduce the behavior, 
1. `ros_cross_compile /path/to/my/workspace --arch aarch64 --os ubuntu --rosdistro dashing`

## System 
 - OS: [Ubuntu 18.04 Aarch64]
 - ROS : [melodic]

emersonknapp, 2021-04-22:
Are you able to run docker commands without sudo on this system? E.g.

```
docker run hello-world
```

casmer, 2021-04-25:
I am having this issue, and yes, docker hello world works fine. Is there a way to find out which url its trying to access when the error occurs?

malith1992, 2021-04-26:
> Are you able to run docker commands without sudo on this system? E.g.
> 
> ```
> docker run hello-world
> ```
> 
> If not, you'll need to add your user to the `docker` group

After adding the user to the docker group, the permission denied issue got fixed. But now I have another issue while running `ros_cross_compile /home/nvidia/catkin_ws --arch aarch64 --os ubuntu --rosdistro dashing`

>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/docker_client.py", line 145, in run_container
>     image_name, exit_code, '', image_name, 'See above ^')
> docker.errors.ContainerError: Command '' in image 'ros_cross_compile:rosdep' returned non-zero exit status 1: See above ^
> 

I here by attach the last part of the terminal log

> INFO:Docker Client:Step 6/10 : RUN rosdep init
> INFO:Docker Client: ---> Running in 500394eccb65
> INFO:Docker Client:Wrote /etc/ros/rosdep/sources.list.d/20-default.list
> Recommended: please run
> 
> 	rosdep update
> INFO:Docker Client: ---> 6248cc1e9b2f
> INFO:Docker Client:Step 7/10 : COPY gather_rosdeps.sh /root/
> INFO:Docker Client: ---> 08547f543c8b
> INFO:Docker Client:Step 8/10 : RUN mkdir -p /ws
> INFO:Docker Client: ---> Running in 23865e801aec
> INFO:Docker Client: ---> c1bf0d3a603c
> INFO:Docker Client:Step 9/10 : WORKDIR /ws
> INFO:Docker Client: ---> Running in 2b4778c06d38
> INFO:Docker Client: ---> 42e20a74f5ce
> INFO:Docker Client:Step 10/10 : ENTRYPOINT ["/root/gather_rosdeps.sh"]
> INFO:Docker Client: ---> Running in 5db4ea6fbee6
> INFO:Docker Client: ---> 2356cbd76854
> INFO:Docker Client:Successfully built 2356cbd76854
> INFO:Docker Client:Successfully tagged ros_cross_compile:rosdep
> INFO:Rosdep Gatherer:Running rosdep collector image on workspace /home/nvidia/catkin_ws
> INFO:Docker Client:+ '[' '!' -d ./src ']'
> INFO:Docker Client:+ '[' -f /usercustom/rosdep_setup ']'
> INFO:Docker Client:++ dirname cc_internals/aarch64-ubuntu-dashing/install_rosdeps.sh
> INFO:Docker Client:+ out_dir=cc_internals/aarch64-ubuntu-dashing
> INFO:Docker Client:+ mkdir -p cc_internals/aarch64-ubuntu-dashing
> INFO:Docker Client:+ rosdep update
> INFO:Docker Client:Warning: running 'rosdep update' as root is not recommended.
> INFO:Docker Client:  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
> INFO:Docker Client:reading in sources list data from /etc/ros/rosdep/sources.list.d
> INFO:Docker Client:Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
> INFO:Docker Client:Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
> INFO:Docker Client:Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
> INFO:Docker Client:Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
> INFO:Docker Client:Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
> INFO:Docker Client:Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
> INFO:Docker Client:Skip end-of-life distro "ardent"
> INFO:Docker Client:Skip end-of-life distro "bouncy"
> INFO:Docker Client:Skip end-of-life distro "crystal"
> INFO:Docker Client:Add distro "dashing"
> INFO:Docker Client:Skip end-of-life distro "eloquent"
> INFO:Docker Client:Add distro "foxy"
> INFO:Docker Client:Add distro "galactic"
> INFO:Docker Client:Skip end-of-life distro "groovy"
> INFO:Docker Client:Skip end-of-life distro "hydro"
> INFO:Docker Client:Skip end-of-life distro "indigo"
> INFO:Docker Client:Skip end-of-life distro "jade"
> INFO:Docker Client:Add distro "kinetic"
> INFO:Docker Client:Skip end-of-life distro "lunar"
> INFO:Docker Client:Add distro "melodic"
> INFO:Docker Client:Add distro "noetic"
> INFO:Docker Client:Add distro "rolling"
> INFO:Docker Client:updated cache in /root/.ros/rosdep/sources.cache
> INFO:Docker Client:+ cat
> INFO:Docker Client:+ mapfile -t package_paths
> INFO:Docker Client:++ colcon list -p
> INFO:Docker Client:+ rosdep install --os ubuntu:bionic --rosdistro dashing --from-paths src/beginner_tutorials src/ddynamic_reconfigure src/realsense-ros/realsense2_camera src/realsense-ros/realsense2_description --ignore-src --reinstall --default-yes --skip-keys '' --simulate
> INFO:Docker Client:ERROR: the following packages/stacks could not have their rosdep keys resolved
> INFO:Docker Client:to system dependencies:
> INFO:Docker Client:beginner_tutorials: Cannot locate rosdep definition for [message_runtime]
> INFO:Docker Client:ddynamic_reconfigure: Cannot locate rosdep definition for [rostest]
> INFO:Docker Client:realsense2_camera: Cannot locate rosdep definition for [tf]
> INFO:Docker Client:realsense2_description: Cannot locate rosdep definition for [rosunit]
> Traceback (most recent call last):
>   File "/home/nvidia/.local/bin/ros_cross_compile", line 11, in <module>
>     sys.exit(main())
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/ros_cross_compile.py", line 227, in main
>     cross_compile_pipeline(args, data_collector, platform)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/ros_cross_compile.py", line 214, in cross_compile_pipeline
>     stage(platform, docker_client, ros_workspace_dir, options, data_collector)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/dependencies.py", line 136, in __call__
>     custom_data_dir=options.custom_data_dir)
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/dependencies.py", line 87, in gather_rosdeps
>     volumes=volumes,
>   File "/home/nvidia/.local/lib/python3.6/site-packages/ros_cross_compile/docker_client.py", line 145, in run_container
>     image_name, exit_code, '', image_name, 'See above ^')
> docker.errors.ContainerError: Command '' in image 'ros_cross_compile:rosdep' returned non-zero exit status 1: See above ^
>

emersonknapp, 2021-04-26:
`--rosdistro dashing` - You are building for ROS 2 

From reading the output, you are depending on the following packages:
* `message_runtime`
* `rostest`
* `tf`
* `rosunit`

These are not [packages in Dashing](https://github.com/ros/rosdistro/blob/master/dashing/distribution.yaml). You said in your original bug report that you are building for Melodic. Which one is your intention?

malith1992, 2021-04-30:
> `--rosdistro dashing` - You are building for ROS 2
> 
> From reading the output, you are depending on the following packages:
> 
> * `message_runtime`
> * `rostest`
> * `tf`
> * `rosunit`
> 
> These are not [packages in Dashing](https://github.com/ros/rosdistro/blob/master/dashing/distribution.yaml), they are ROS 1 packages. You said in your original bug report that you are building for Melodic. Which one is your intention?

I'm sorry I mistakenly run it for dashing. But when I use melodic, it works fine. Thanks

closed 2021-04-30 by malith1992