ros-tooling/cross_compile #313: Cannot run in gitlab-ci with docker executor: echo 'No src/ directory found at /ws, did you remember to mount your workspace?'

kjeremy, 2021-05-20:
## Description
I am attempting to run this script to cross-compile for melodic. At the moment it fails accessing the workspace.

### Expected Behavior
It should succeed.

### Actual Behavior
It fails:

```
$ ros_cross_compile ros_ws --arch aarch64 --os ubuntu --rosdistro melodic --custom-setup-script ./setup.sh  --skip-rosdep-keys delphi_esr
INFO:Rosdep Gatherer:Building rosdep collector image: ros_cross_compile:rosdep
INFO:Docker Client:Step 1/10 : FROM ubuntu:bionic
INFO:Docker Client: ---> 81bcf752ac3d
INFO:Docker Client:Step 2/10 : RUN apt-get update && apt-get install --no-install-recommends -y       dirmngr       gnupg2     && rm -rf /var/lib/apt/lists/*
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 66e71fc01496
INFO:Docker Client:Step 3/10 : RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 56b494134900
INFO:Docker Client:Step 4/10 : RUN echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 8fa12f0d39a2
INFO:Docker Client:Step 5/10 : RUN apt-get update && apt-get install --no-install-recommends -y       python-rosdep       python3-colcon-common-extensions     && rm -rf /var/lib/apt/lists/*
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 4b3a4f918702
INFO:Docker Client:Step 6/10 : RUN rosdep init
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> d760a8a8591e
INFO:Docker Client:Step 7/10 : COPY gather_rosdeps.sh /root/
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 77fee945736d
INFO:Docker Client:Step 8/10 : RUN mkdir -p /ws
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> d7c12a2077c3
INFO:Docker Client:Step 9/10 : WORKDIR /ws
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> e5eba9dc0d20
INFO:Docker Client:Step 10/10 : ENTRYPOINT ["/root/gather_rosdeps.sh"]
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 217a5be9d9fe
INFO:Docker Client:Successfully built 217a5be9d9fe
INFO:Docker Client:Successfully tagged ros_cross_compile:rosdep
INFO:Rosdep Gatherer:Running rosdep collector image on workspace /builds/ros_ws
INFO:Docker Client:+ '[' '!' -d ./src ']'
INFO:Docker Client:++ pwd
INFO:Docker Client:+ echo 'No src/ directory found at /ws, did you remember to mount your workspace?'
INFO:Docker Client:+ exit 1
INFO:Docker Client:No src/ directory found at /ws, did you remember to mount your workspace?
Traceback (most recent call last):
  File "/usr/local/bin/ros_cross_compile", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/ros_cross_compile.py", line 227, in main
    cross_compile_pipeline(args, data_collector, platform)
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/ros_cross_compile.py", line 214, in cross_compile_pipeline
    stage(platform, docker_client, ros_workspace_dir, options, data_collector)
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/dependencies.py", line 136, in __call__
    custom_data_dir=options.custom_data_dir)
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/dependencies.py", line 87, in gather_rosdeps
    volumes=volumes,
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/docker_client.py", line 145, in run_container
    image_name, exit_code, '', image_name, 'See above ^')
docker.errors.ContainerError: Command '' in image 'ros_cross_compile:rosdep' returned non-zero exit status 1: See above ^
```

## To Reproduce
** Steps to reproduce the behavior, e.g.
Create a gitlab runner with /var/run/docker.sock mounted. In the gitlab.yaml:

```yaml
[... 462 lines cut ...]
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/ros_cross_compile.py", line 227, in main
    cross_compile_pipeline(args, data_collector, platform)
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/ros_cross_compile.py", line 214, in cross_compile_pipeline
    stage(platform, docker_client, ros_workspace_dir, options, data_collector)
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/dependencies.py", line 136, in __call__
    custom_data_dir=options.custom_data_dir)
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/dependencies.py", line 87, in gather_rosdeps
    volumes=volumes,
  File "/usr/local/lib/python3.6/dist-packages/ros_cross_compile/docker_client.py", line 145, in run_container
    image_name, exit_code, '', image_name, 'See above ^')
docker.errors.ContainerError: Command '' in image 'ros_cross_compile:rosdep' returned non-zero exit status 1: See above ^
Cleaning up file based variables 00:01
ERROR: Job failed: exit code 1
```

Notice that in the output of the `tree` command there is a `src/` directory on the workspace.

## System (please complete the following information)
 - OS: Ubuntu Bionic (running in docker)
 - ROS: Melodic

** For ROS 2 bugs - please attach a VCS.repos or Dockerfile if possible to help us reproduce the environment **

## Additional context
** Add any other context about the problem here **

emersonknapp, 2021-05-20:
Am I reading correctly that you're running this from inside a Docker container? The tool itself uses Docker - so, if so, I think there's probably something going wrong with the Docker-in-Docker situation.

kjeremy, 2021-05-20:
Yes that's correct that it should be using docker in docker. I'm not
terribly familiar with docker in docker so am unsure how to debug.

emersonknapp, 2021-05-20:
Are you able to run in the host environment instead? I'm inclined to say "running ros_cross_compile from inside a container is not supported"

kjeremy, 2021-05-20:
Unfortunately I cannot run in the host environment for production. After reading some stack overflow (https://stackoverflow.com/questions/50971417/docker-inside-docker-volume-is-mounted-but-empty) I think a solution may be to use named volumes for the workspace.

emersonknapp, 2021-05-20:
I would be happy to review changes along these lines. However, I think you would have to create the named volume ahead of time - still from the host environment, could you do that once you're already in a container?

kjeremy, 2021-05-21:
I got it working successfully using docker-in-docker with the following:

```yaml
image: teracy/ubuntu:18.04-dind-19.03.3

services:
  - docker:19.03.3-dind

variables:
  # Disable TLS or we get SSLv1 errors. We shouldn't need this since we mount the /certs volume.
  DOCKER_TLS_CERTDIR: ""
  DOCKER_HOST: tcp://docker:2375
```

While it would be nice to support with the "sharing the docker socket approach" this is a kind of workaround.

kjeremy, 2021-05-25:
I'm going to close this since it works in dind.

closed 2021-05-25 by kjeremy

emersonknapp, 2021-05-25:
Others may appreciate your findings as a section in the README - if you're willing to contribute that. Maybe something like a "Troubleshooting" section, under which there is a subsection for "No src/ directory found at /ws, did you remember to mount your workspace?"

ZhamaoDra, 2021-06-28:
hi, is there more instruction？

kjeremy, 2021-06-28:
@ZhamaoDra What I did was set up a gitlab-runner running in docker in privileged mode and used the .gitlab-ci.yml file in the troubleshooting section (#314). I'm not sure what else you are looking for.