ros-tooling/cross_compile #327: cross compile worked a few months ago is no longer working

casmer, 2021-06-14:
I am running the command 
ros_cross_compile /home/caseyg/ros_dev --arch armhf --os ubuntu --rosdistro dashing 
and during the setup process i get the errors 
INFO:Docker Client:lalosoft_robot_shared: Cannot locate rosdep definition for [ament_cmake]
INFO:Docker Client:lalosoft_robot_jazzy_drive_host: Cannot locate rosdep definition for [std_msgs]
INFO:Docker Client:lalosoft_robot_control_host: Cannot locate rosdep definition for [sensor_msgs]
INFO:Docker Client:lalosoft_robot_drive_host: Cannot locate rosdep definition for [std_msgs]
INFO:Docker Client:lalosoft_robot_msgs: Cannot locate rosdep definition for [rosidl_default_runtime]

but all of these definitions have worked the last time i built this code base. Am I missing something?

INFO:Docker Client:+ rosdep install --os ubuntu:bionic --rosdistro dashing --from-paths src/ros2robot/robot_control_host src/ros2robot/robot_drive_host src/ros2robot/robot_jazzy_drive_host src/ros2robot/robot_msgs src/ros2robot/robot_shared --ignore-src --reinstall --default-yes --skip-keys 'robot_msgs robot_shared' --simulate
INFO:Docker Client:WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 2
INFO:Docker Client:ERROR: the following packages/stacks could not have their rosdep keys resolved
INFO:Docker Client:to system dependencies:
INFO:Docker Client:lalosoft_robot_shared: Cannot locate rosdep definition for [ament_cmake]
INFO:Docker Client:lalosoft_robot_jazzy_drive_host: Cannot locate rosdep definition for [std_msgs]
INFO:Docker Client:lalosoft_robot_control_host: Cannot locate rosdep definition for [sensor_msgs]
INFO:Docker Client:lalosoft_robot_drive_host: Cannot locate rosdep definition for [std_msgs]
INFO:Docker Client:lalosoft_robot_msgs: Cannot locate rosdep definition for [rosidl_default_runtime]
Traceback (most recent call last):
  File "/home/caseyg/.local/bin/ros_cross_compile", line 8, in <module>
    sys.exit(main())
  File "/home/caseyg/.local/lib/python3.8/site-packages/ros_cross_compile/ros_cross_compile.py", line 227, in main
    cross_compile_pipeline(args, data_collector, platform)
  File "/home/caseyg/.local/lib/python3.8/site-packages/ros_cross_compile/ros_cross_compile.py", line 214, in cross_compile_pipeline
    stage(platform, docker_client, ros_workspace_dir, options, data_collector)
  File "/home/caseyg/.local/lib/python3.8/site-packages/ros_cross_compile/dependencies.py", line 130, in __call__
    gather_rosdeps(
  File "/home/caseyg/.local/lib/python3.8/site-packages/ros_cross_compile/dependencies.py", line 77, in gather_rosdeps
    docker_client.run_container(
  File "/home/caseyg/.local/lib/python3.8/site-packages/ros_cross_compile/docker_client.py", line 144, in run_container
    raise docker.errors.ContainerError(
docker.errors.ContainerError: Command '' in image 'ros_cross_compile:rosdep' returned non-zero exit status 1: See above ^

christophebedard, 2021-06-14:
Dashing has just recently been marked as end-of-life (https://github.com/ros/rosdistro/pull/29845), so by default `rosdep` won't consider it. The `--include-eol-distros` option for `rosdep update` must be used to use/consider EOL distros (https://github.com/ros-infrastructure/rosdep/pull/647).

However, it seems like ros_cross_compile already does: https://github.com/ros-tooling/cross_compile/pull/283

What version of ros_cross_compile are you using?

casmer, 2021-06-15:
Thank you, i was running 0.6, removed it with pip3, and re-installed to get 0.8, its working now. thank you so much!

closed 2021-06-15 by casmer