ros-tooling/cross_compile #328: Error building due to ROS GPG key unavailable

sanatmharolkar, 2021-06-21:
The ROS packages public key F42ED6FBAB17C654 is no longer available due to a recent incident, it has been replaced with a new key. The package ros_cross_compile is still trying to download the packages in Docker using the old key. The build is failing due to this.

TSC21, 2021-06-21:
That was already fixed in https://github.com/ros-tooling/cross_compile/pull/317. What version of thd tool are you using?

sanatmharolkar, 2021-06-21:
I installed using "pip3 install ros_cross_compile". It is version 0.8.0

TSC21, 2021-06-21:
The above PR changes already include the update to whwre the keys are being fetched from. What ROS distro, arch and OS are you trying this against?

sanatmharolkar, 2021-06-21:
**Target:**
OS - Debian (Buster)
Arch - armhf
Distro - Foxy

**Host:**
OS - Ubuntu 20.04
Arch - amd64

TSC21, 2021-06-21:
> **Target:**
> OS - Debian (Buster)
> Arch - armhf
> Distro - Foxy
> 
> **Host:**
> OS - Ubuntu 20.04
> Arch - amd64

Foxy is not available in Debian Buster for `armhf` - https://github.com/ros-tooling/cross_compile/blob/master/test/run_e2e_test.sh#L91.

sanatmharolkar, 2021-06-21:
Oh. I guess I'll have to do the compilation manually. Thanks.

closed 2021-06-21 by sanatmharolkar