ros-tooling/cross_compile #340: Cross-compilation for armhf fails due to "Unable to locate package"

Sanjaer, 2021-09-14:
## Description
No problem compiling for aarch64 nor x86_64. However when cross-compiling for armhf no packets are located via apt, so compilation fails.

This is a snapsot of the error:
`INFO:Docker Client:Unable to locate package ros-galactic-std-srvs`

And this is the full error:

```INFO:Docker Client:Step 22/31 : RUN export DEBIAN_FRONTEND=noninteractive && apt-get update &&     ./install_rosdeps.sh &&     rm -rf /var/lib/apt/lists/*
INFO:Docker Client: ---> [Warning] The requested image's platform (linux/arm/v7) does not match the detected host platform (linux/amd64) and no specific platform was requested
INFO:Docker Client: ---> Running in b8e8859e8447
INFO:Docker Client:Get:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease [265 kB]
INFO:Docker Client:Get:2 http://packages.ros.org/ros2/ubuntu focal InRelease [4670 B]
INFO:Docker Client:Get:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
INFO:Docker Client:Get:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [101 kB]
INFO:Docker Client:Get:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
INFO:Docker Client:Get:6 http://packages.ros.org/ros2/ubuntu focal/main armhf Packages [38.9 kB]
INFO:Docker Client:Get:7 http://ports.ubuntu.com/ubuntu-ports focal/main armhf Packages [1227 kB]
INFO:Docker Client:Get:8 http://ports.ubuntu.com/ubuntu-ports focal/restricted armhf Packages [10.8 kB]
INFO:Docker Client:Get:9 http://ports.ubuntu.com/ubuntu-ports focal/multiverse armhf Packages [141 kB]
INFO:Docker Client:Get:10 http://ports.ubuntu.com/ubuntu-ports focal/universe armhf Packages [10.9 MB]
INFO:Docker Client:Get:11 http://ports.ubuntu.com/ubuntu-ports focal-updates/main armhf Packages [1012 kB]
INFO:Docker Client:Get:12 http://ports.ubuntu.com/ubuntu-ports focal-updates/multiverse armhf Packages [9298 B]
INFO:Docker Client:Get:13 http://ports.ubuntu.com/ubuntu-ports focal-updates/universe armhf Packages [869 kB]
INFO:Docker Client:Get:14 http://ports.ubuntu.com/ubuntu-ports focal-updates/restricted armhf Packages [15.6 kB]
INFO:Docker Client:Get:15 http://ports.ubuntu.com/ubuntu-ports focal-backports/universe armhf Packages [6327 B]
INFO:Docker Client:Get:16 http://ports.ubuntu.com/ubuntu-ports focal-backports/main armhf Packages [2679 B]
INFO:Docker Client:Get:17 http://ports.ubuntu.com/ubuntu-ports focal-security/restricted armhf Packages [15.3 kB]
INFO:Docker Client:Get:18 http://ports.ubuntu.com/ubuntu-ports focal-security/main armhf Packages [611 kB]
INFO:Docker Client:Get:19 http://ports.ubuntu.com/ubuntu-ports focal-security/universe armhf Packages [598 kB]
INFO:Docker Client:Get:20 http://ports.ubuntu.com/ubuntu-ports focal-security/multiverse armhf Packages [4719 B]
INFO:Docker Client:Fetched 16.1 MB in 33s (494 kB/s)
Reading package lists...
INFO:Docker Client:+ apt-get install -y qt5-qmake ros-galactic-ament-lint-auto ros-galactic-std-srvs ros-galactic-rclcpp ros-galactic-std-msgs ros-galactic-ament-index-cpp ros-galactic-rosidl-default-runtime ros-galactic-ament-cmake libqt5gui5 ros-galactic-geometry-msgs ros-galactic-rclcpp-action ros-galactic-ament-lint-common qtbase5-dev ros-galactic-rosidl-default-generators libqt5core5a

INFO:Docker Client:Reading package lists...
INFO:Docker Client:Building dependency tree...
INFO:Docker Client:Reading state information...
INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-ament-lint-auto
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-std-srvs
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-rclcpp
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-std-msgs
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-ament-index-cpp
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-rosidl-default-runtime
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-ament-cmake
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-geometry-msgs
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-rclcpp-action
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-ament-lint-common
INFO:Docker Client:

INFO:Docker Client:E
INFO:Docker Client:: 
INFO:Docker Client:Unable to locate package ros-galactic-rosidl-default-generators
INFO:Docker Client:

ERROR:Docker Client:Error building Docker image. The follow error was caught:
The command '/bin/bash -c export DEBIAN_FRONTEND=noninteractive && apt-get update &&     ./install_rosdeps.sh &&     rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
NoneType: None
Traceback (most recent call last):
  File "/home/mobuser/.local/bin/ros_cross_compile", line 8, in <module>
    sys.exit(main())
  File "/home/mobuser/.local/lib/python3.8/site-packages/ros_cross_compile/ros_cross_compile.py", line 237, in main
    cross_compile_pipeline(args, data_collector, platform)
  File "/home/mobuser/.local/lib/python3.8/site-packages/ros_cross_compile/ros_cross_compile.py", line 224, in cross_compile_pipeline
    stage(platform, docker_client, ros_workspace_dir, options, data_collector)
  File "/home/mobuser/.local/lib/python3.8/site-packages/ros_cross_compile/sysroot_creator.py", line 172, in __call__
    create_workspace_sysroot_image(docker_client, platform)
  File "/home/mobuser/.local/lib/python3.8/site-packages/ros_cross_compile/sysroot_creator.py", line 143, in create_workspace_sysroot_image
    docker_client.build_image(
  File "/home/mobuser/.local/lib/python3.8/site-packages/ros_cross_compile/docker_client.py", line 73, in build_image
    self._process_build_log(log_generator)
  File "/home/mobuser/.local/lib/python3.8/site-packages/ros_cross_compile/docker_client.py", line 83, in _process_build_log
    raise docker.errors.BuildError(error_line)
docker.errors.BuildError: The command '/bin/bash -c export DEBIAN_FRONTEND=noninteractive && apt-get update &&     ./install_rosdeps.sh &&     rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100```

## To Reproduce
Simply try cross-compiling for armhf

## System (please complete the following information)
 - OS:Ubuntu Focal
 - ROS 2 Distro: Galactic

As I said, this works for aarch64 and x86_64

christophebedard, 2021-09-14:
There are no binary packages for armhf, so that's expected: https://www.ros.org/reps/rep-2000.html#galactic-geochelone-may-2021-november-2022.

You'll probably need to have everything built/cross-compiled instead of relying on ROS 2 core binaries.

emersonknapp, 2021-09-14:
That's right - you'll have to build all of [ros2.repos](https://github.com/ros2/ros2/blob/master/ros2.repos). There is no current guarantee that this build will work, because `armhf` is not a supported-Tier platform for ROS 2 in recent distributions, so the code may not be properly portable anymore.

closed 2021-09-14 by emersonknapp

HeinzBenjamin, 2022-04-05:
> That's right - you'll have to build all of [ros2.repos](https://github.com/ros2/ros2/blob/master/ros2.repos). There is no current guarantee that this build will work, because `armhf` is not a [...]

Could you kindly elaborate on how this can be done?
Cheers

emersonknapp, 2022-05-19:
```
git clone https://github.com/ros2/ros2
cd ros2
mkdir src
vcs import src < ros2.repos
ros-cross-compile .  # etc options
```