ros-tooling/cross_compile #342: /share directory is not created after ros_cross_compile

spitfire1989, 2021-11-02:
Hi, I am using the tool in a VM running Ubuntu 18.04.04, after the [installation](https://github.com/ros-tooling/cross_compile#installation) (I install the latest Docker engine), I follow the [tutorial](https://github.com/ros-tooling/cross_compile#tutorial) but after the ros_cross_compile command I realize that:

1.- tutorial.repos is not created in my workspace
2.- the path install_aarch64/share is unreachable because the share directory is not created, I realize this when I copy the directory install_aarch64 to my target device and run 

rosdep install --from-paths install_aarch64/share --ignore-src --rosdistro foxy -y

I got an error that says the path install_aarch64/share is not found

What could be the problem?

emersonknapp, 2021-11-03:
1. `tutorial.repos` is left over from an old version of the tutorial, this is not something that the build should create. I have removed it from the list of files in the README.
2. Thanks for reporting this! The `install/share` directory will only be there if the build was created with `--merge-install`. Otherwise, each individual built package will have its own `install/PACKAGE/share` directory. For this portion of the tutorial, I would recommend adding merge install to the defaults.yaml. It's easier to use merge-installed colcon outputs across machines, so this is probably best practice for a deployable artifact anyways.

I am also noticing that a bug was introduced in the 0.9 version so that a `defaults.yaml` in the workspace directory will not be respected unless explicitly specified.

I have updatd the README in https://github.com/ros-tooling/cross_compile/commit/921884ef5eca9017df390d9e664ce4e1d83e5316 to have the correct commands necessary for the tutorial to work as expected at the moment, and will try to put in a fix for the `defaults.yaml` thing shortly.

spitfire1989, 2021-11-04:
@emersonknapp Hi, did you test this with the target platform?, now the share directory is created but when I run ` rosdep install --from-paths install_aarch64/share --ignore-src --rosdistro foxy -y` I got an error:

E: Unable to locate package ros-foxy-example-interfaces
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-foxy-example-interfaces] failed

My target is a custom Linux distribution from Xilinx built from Yocto, its name is Petalinux, and I installed Ubuntu 20.04 LTS over that distro

emersonknapp, 2021-11-04:
Assuming you have APT on this platform and have ROS dependencies you need to install that way, you still need to set up the ROS 2 APT sources as you would for any system, as described at https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html#setup-sources. I will add some language making this clear.

spitfire1989, 2021-11-04:
I already try that, I followed this steps:
[add the repo](https://docs.ros.org/en/foxy/Installation/Ubuntu-Development-Setup.html#id4), [even I installed the development tools](https://docs.ros.org/en/foxy/Installation/Ubuntu-Development-Setup.html#id5), and I get the error I mentioned before

emersonknapp, 2021-11-04:
You shouldn't need the development tools - that's for building, which you have already done.

I wonder if maybe the sources setup didn't work correctly for your custom distro. 

* Are you able to `sudo apt-cache search ros-foxy` - does this show any packages at all?
* What are the contents of `/etc/apt/sources.list.d/ros2.list`?
* Are there any errors when you `apt-get update`?

spitfire1989, 2021-11-08:
hi @emersonknapp I could solve the error:

**E: Unable to locate package ros-foxy-example-interfaces
ERROR: the following rosdeps failed to install
apt: command [sudo -H apt-get install -y ros-foxy-example-interfaces] failed**

it was my fault, I was using a 32 bits ubuntu distro in my embedded linux instead a 64 bits, once I run `rosdep install --from-paths install_aarch64/share --ignore-src --rosdistro foxy -y` a message says that all rosdep dependencies were installed, then I write this commands:

`source install_aarch64/setup.bash
ros2 run demo_nodes_cpp talker`

but now I get:
**-bash: ros2: command not found**

I will answer your questions:
1.- Are you able to sudo apt-cache search ros-foxy - does this show any packages at all?
[image]

2.- What are the contents of /etc/apt/sources.list.d/ros2.list?
[image]

3.- Are there any errors when you apt-get update?
no, I can make updates, upgrades and install packages without issues

spitfire1989, 2021-11-08:
"Assuming you have APT on this platform and have ROS dependencies you need to install that way, you still need to set up the ROS 2 APT sources as you would for any system"

What dependencies are you talking about?

spitfire1989, 2021-11-09:
I used the `sudo apt install ros-foxy-ros-base` and then works!, but that step is not in your tutorial

spitfire1989, 2021-11-11:
@emersonknapp when testing the demo in the target board you must add the sources for the listener too: 

source install_aarch64/setup.bash
ros2 run demo_nodes_cpp talker

# and in a different shell, add sources too
source install_aarch64/setup.bash 
ros2 run demo_nodes_cpp listener