ros-tooling/cross_compile #344: Error when building sysroot image for arm64v8/ubuntu:focal

petternorgren, 2021-11-18:
## Problem
I run the command ros_cross_compile . --arch aarch64 --os ubuntu --rosdistro galactic and I get the error:

ERROR:Docker Client:Error building Docker image. The follow error was caught:
The command '/bin/bash -c echo 'Etc/UTC' > /etc/timezone &&     ln -sf /usr/share/zoneinfo/Etc/UTC /etc/localtime' returned a non-zero code: 127

I have tried changing the sysroot docker image file and from what I can tell it seems that the issue occurs since it can't find /bin/bash which seems odd.

Not sure how to proceed with this error. Does anybody have any suggestions?

## Logs
Attached is the terminal output (for the part building the sysroot image):

INFO:ros_cross_compile.sysroot_creator:Building sysroot image: pn/aarch64-ubuntu-galactic:latest
INFO:Docker Client:Step 1/31 : ARG BASE_IMAGE
INFO:Docker Client:Step 2/31 : FROM ${BASE_IMAGE}
INFO:Docker Client: ---> d5ca7a445605
INFO:Docker Client:Step 3/31 : ARG ROS_VERSION
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> d0eb7929425b
INFO:Docker Client:Step 4/31 : SHELL ["/bin/bash", "-c"]
INFO:Docker Client: ---> Using cache
INFO:Docker Client: ---> 9b525fb389d7
INFO:Docker Client:Step 5/31 : COPY bin/* /usr/bin/
INFO:Docker Client: ---> d190484ede0b
INFO:Docker Client:Step 6/31 : RUN echo 'Etc/UTC' > /etc/timezone &&     ln -sf /usr/share/zoneinfo/Etc/UTC /etc/localtime
INFO:Docker Client: ---> [Warning] The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/amd64) and no specific platform was requested
INFO:Docker Client: ---> Running in d3f12df5c0fb
INFO:Docker Client:/bin/bash: echo 'Etc/UTC' > /etc/timezone &&     ln -sf /usr/share/zoneinfo/Etc/UTC /etc/localtime: No such file or directory

ERROR:Docker Client:Error building Docker image. The follow error was caught:
The command '/bin/bash -c echo 'Etc/UTC' > /etc/timezone &&     ln -sf /usr/share/zoneinfo/Etc/UTC /etc/localtime' returned a non-zero code: 127
NoneType: None
Traceback (most recent call last):
  File "/usr/local/bin/ros_cross_compile", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/ros_cross_compile/ros_cross_compile.py", line 237, in main
    cross_compile_pipeline(args, data_collector, platform)
  File "/usr/local/lib/python3.9/dist-packages/ros_cross_compile/ros_cross_compile.py", line 224, in cross_compile_pipeline
    stage(platform, docker_client, ros_workspace_dir, options, data_collector)
  File "/usr/local/lib/python3.9/dist-packages/ros_cross_compile/sysroot_creator.py", line 172, in __call__
    create_workspace_sysroot_image(docker_client, platform)
  File "/usr/local/lib/python3.9/dist-packages/ros_cross_compile/sysroot_creator.py", line 143, in create_workspace_sysroot_image
    docker_client.build_image(
  File "/usr/local/lib/python3.9/dist-packages/ros_cross_compile/docker_client.py", line 73, in build_image
    self._process_build_log(log_generator)
  File "/usr/local/lib/python3.9/dist-packages/ros_cross_compile/docker_client.py", line 83, in _process_build_log
    raise docker.errors.BuildError(error_line)
docker.errors.BuildError: The command '/bin/bash -c echo 'Etc/UTC' > /etc/timezone &&     ln -sf /usr/share/zoneinfo/Etc/UTC /etc/localtime' returned a non-zero code: 127

## System
 - OS: Ubuntu 21.10
 - ROS 2 Distro: Galactic
 - Docker version: 20.10.11
 - Python version: 3.9.7

emersonknapp, 2021-12-01:
Do you know what the value of BASE_IMAGE resolves to?

davidbro-in, 2022-03-18:
Hello @petternorgren do you find a solution? I have the same problem

davidbro-in, 2022-03-18:
For me, it solves the problem `docker run --rm --privileged multiarch/qemu-user-static --reset -p yes`. But I do not know what it does.